stefanrueger/urboot #1: To support mega0 AVR and Tiny0/1/2 AVR, similar to optiboot_x

mcuee, 2022-11-12:
I am not so sure how difficult it is to add the support of mega0 AVR and Tiny0/1/2 AVR, similar to optiboot_x.

Stretch goal will be to add the support of AVR DA/DB/DD part, similar to optiboot_dx.

Ref: optiboot_dx is too diffierent to be included in optiboot (optiboot_x is integrated into optiboot).
* https://github.com/Optiboot/optiboot/issues/353

stefanrueger, 2022-11-15:
My understanding is that optiboot_dx doesn't work. I'd be interested in handling XMEGA parts (They have a huge bootloader section that I think could be made available for ordinary applications) and perhaps extend to the parts that optiboot_x can handle.

Immediate priority though is to deal well with classic parts and release a reference implementation (current code is pre-release).

mcuee, 2022-11-15:
> My understanding is that optiboot_dx doesn't work

optiboot_dx works well with avrdude on AVR DA/DB parts based on testing on the 128KB Flash parts. I believe it should also work for 64KB and less Flash parts (pending conformation_. 

optiboot_dx does not work yet for AVR DD parts.

mcuee, 2022-11-15:
> I'd be interested in handling XMEGA parts (They have a huge bootloader section that I think could be made available for ordinary applications) and perhaps extend to the parts that optiboot_x can handle.

I have seen any STK500v1 implementation for XMEGA parts yet. The popular bootloaders for XMEGA part are AVR109 protocol (eg: xboot).

xboot: https://github.com/alexforencich/xboot

SpenceKonde, 2022-12-20:
Dx and non-Dx modern AVRs use fundamentally different methods to write the flash.

On non-Dx, you fill a page buffer and then execute page-erase-write command. 

On Dx, you have to have erased the pages, and have NVMCMD set to FLWR but then just load low and high byte of word into R0 and R1, have the Z pointer pointing to where you want to write and execute SPM or SPM Z+.  (You can also use the mapping and normal st instructions, but this looked like it would take more flash, is twice as slow (takes same time to write a byte as a word) and doesn't work correctly on the AVR128DA due to an erratum. Erasing pages is done by selecting an FLPER NVMCMD, setting the Z pointer to point to the page being erased, 

All modern AVRs also have the following differences:
They all have hardware bootloader support.
They have three memory sections, which are located in this order in the flash:
* BOOTCODE
* APPCODE
* APPDATA
There are two fuses which set the size of these sections
BOOTSIZE/BOOTEND: size of bootloader in chunks of size dependant on device (currently, 256 for all released parts except Dx, where it's 512
CODESIZE/APPEND: Size of bootloader + appcode sections, in above described chunks. The original name was APPEND, (for APPlication END. For obvious reasons, that was a bad naming choice, and they have switched to BOOTSIZE and CODESIZE for new parts. 

Code executing from APPDATA cannot write to any NVM, not even EEPROM.
Code executing from APPCODE can write to EEPROM and APPDATA
Code executing from BOOTCODE can write to EEPROM, APPDATA, and APPCODE. 
Nothing except UPDI can write to BOOTCODE section. A bootloader can never update or overwrite itself. 
If the bootloader section size is 0, the entire flash is BOOTCODE.
It is not specified in the datasheet what can write to USERROW - likely everything except APPDATA. 
Future parts will also contain a "BOOTROW", generally of the same size as the USERROW.
The function of this section has yet to be made clear. 
Code executing from the bootloader section can set the BOOTRP bit, which takes effect when execution crosses to the application section. Setting this bit will cause all data in the boot section to read as 0x0000 (nop). This includes instruction fetch, so if it has been set you can't call into the bootloader to write to APPCODE - all that it'll see there are nop's. There is also an APPCODEWP and APPDATAWP bit, when set, they prevent writes to the specified section. All of these bits (along with FLMAPLOCK) are protected by CCP, and once set, can be cleared only by a reset. 

Bootloaders start at  0x0000, not near the end of the flash. 
 
On Dx, a write for the purposes of protection only counts the SPM or ST that writes or erases flash. 
On other modern AVRs, both filling the page buffer and committing the page buffer must be carried out from the appropriate section.

stefanrueger, 2022-12-20:
@SpenceKonde Thanks for a thorough overview espec with regards to the pitfalls.

The urboot project originated in 2016, but I only recently published it (following a New-Year resolution to give back to the open-source community).

I was hoping that not much needs changing in the bootloader for newer parts over and above changing the way flash/eeprom is written (or read) but may have underestimated some of the sublte differences you pointed out. Anyway, my plan was to first see whether/how urboot bootloaders are accepted along with `avrdude -c urclock` and the [urprotocol](https://github.com/stefanrueger/urboot/blob/main/urprotocol.md) before expanding to non-classic parts.

SpenceKonde, 2022-12-21:
IIRC (it's been a while since I looked at the details of the classic AVR bootloader write process, modern-non-dx probably isn't too hard of a port. You always start up at ~3.33 or ~2.66 MHz, (16/6 or 20/6), which is plenty fast enough for reading from the serial port - because of the fractional baud rate generator, the baud rate calculation error that plagued AVR in the past is gone - so you only have to consider two speeds, and there are no dead-zones between the divisors. You specify the time between the 8 or 16 samples in 64ths of a system clock with the caveat that it must be not less than 64. None of that garbage where 12 MHz chips receive one baud rate more reliably and 16 MHz ones like a different, lower baud rate if you're sticking to "standard" baud rates, and none of this UART crystal nonsense. 57600 and 115200 are some awful baud rates to hit on classic AVRs with a power-of-two-MHz system clock aren't they? Somewhere I wrote up how you could have 4 classic AVRs all running at 115200 baud (or what you think is 115200 baud, with every member of the quarrelsome quartet being able to talk to at least one other device, but none of them successfully able to talk to all three other devices plus a serial adapter. It was after I had to explain to someone on a github issue that no, those classic AVRs running at 8 MHz were not actually operating at 115200 baud, and are only able to talk to eachother, and that no it doesn't matter that they all have a crystal this isn't clock error it's baud error (apparently he  had, for years, had a number of classic AVRs running at 8 MHz and what he thought was 115200 baud (no, it's 111111 baud), and complained that his new tiny 3216 couldn't talk to them and it had to be a problem with my core. Initially he didn't belive me, but he stopped posting when I suggested he could prove this was working by trying to listen in with a serial adapter and noticing that it didn't work (of course it didn't, they're talking at 111111 baud and it's trying to talk at 115108 baud, ie. it's actually able to do what you tell it. 

It's a whole new world in terms of the UART (and for basically no increased effort. The USART baud rate generator is just better. . IIRC flash writing isnt much different to set up, and changed less than you might think. And for non-DX/future EX. everything is accessible in the same address space (fully mapped flash)

mcuee, 2022-12-21:
> DX is definitely more of a challenge.

So much so that @WestfW mentioned below.
* https://github.com/Optiboot/optiboot/issues/353#issuecomment-1312347766
> So ... it turns out that the mega0 and xTiny chips supported by Optiboot_x are substantially different than the new AVR-Dx, and it seems that "fixing" Optiboot_x (or Optiboot) is not a good strategy.
> 
> @SpenceKonde has implemented a new bootloader "Optiboot_dx", based on optiboot, that does support them as part of https://github.com/SpenceKonde/DxCore, and I suspect that will be the final state of things.

stefanrueger, 2022-12-21:
Thanks for the heads up re USART.

| 57600 and 115200 are some awful baud rates to hit on classic AVRs with a power-of-two-MHz system clock aren't they?

When compiled urboot.c warns of baud quantisation errors in excess of 2.5%. The precompiled urboot bootloaders use software IO instead of the USART when the latter has unacceptable quantisation errors. SWIO has less quantisiation error for classic parts at the expense of code space.

stefanrueger, 2023-01-22:
| Dx and non-Dx modern AVRs use fundamentally different methods to write the flash

@SpenceKonde What is a good way for a C preprocessor to distinguish between Dx and non-Dx modern AVRs? Obvs, there is the tedious listing of `#if defined(__AVR_AVR16DD14__) || defined(...)`, but is there a better way to do this?

I have come up with

```
#if defined(FUSE_BOOTEND)    // Non-Dx/Ex modern AVR, BOOTCODE is multiple of 256
...
#elif defined(FUSE_BOOTSIZE) // Dx or Ex modern AVR, BOOTCODE is multiple of 512
...
#endif
```

Any intell on Ex parts and how they need to be differently treated from Dx parts?

SpenceKonde, 2023-01-23:
__AVR_ARCH__ >= 100 for modern AVRs. 

103 indicates a part with fully mapped flash, like a mega0 or a modern tiny. The practical upper limit is 48k of flash as they did on the mega0, with the progmem offset of 0x4000 leaving 3/4ths of the address space as mapped flash, and thus 16k to fit the ram and the special function registers. 
102 indicates a part with 64k flash, hence the flash cannot all be mapped as it would entirely fill 16-bit address space - instead only half (32k) of it is mapped (by default, the upper half), but addresses still fit in 16 bits. If these have a RAMPZ register (the 64k DA/DB do, while the 64k DDs don't have it. 32k DA/DB parts don't have a RAMPZ in the headers) it is vestigial and not used.  I don't think RAMPZ does anything if the parts have it, and while it's in the headers there may not be any corresponding register in the silicon, so it would be an unwritable locations that always reads as 0x00 like most of it's neighbors in the upper 32 bytes of the I/O space. It's unclear what it would do if it is physically present; I've never tried writing to it. It probably "should" remain 0x00 regardless of attempts to write to it, since the unused bits of RAMPZ are supposed to always read as 0, and that's the only kind, 
104 indicates a part with 128k flash, making any location word addressable with a 16-bit address but not necessarily byte addressable, and the memory mapping can map 1/4th of the flash (again, by default this is the last 32k). So the first half of flash can be accessed with LPM, PROGMEM, and so on, and the last 1/4th can be accessed with normal load operations, while the cursed ground between 64k and 96k is accessible only with ELPM, and the RAMPZ register is needed. 

The EA-series looks like it was "forked" from the Dx-series before those parts got any of the big architectural enhancements. They have the same 16/20 selectable base clock instead of the nice directly selectable clock speed, the flash is written and erased by page, and available information thus far suggests that they behave essentially the same as the tinys. There is one big difference at least, though - WWR functionality, so you may have to poll a status register to know when you can start writing the next page. The details of that implementation haven't been released. But what is clear is that the self programming of the EA and EB will have more in common with the tinyavrs and megaavr 0-series.

stefanrueger, 2023-01-23:
Blimey - that's a maze!

My gcc-avr v12.2.0 does not know Dx parts (nor Ex parts). For all the others  I get:
```
$ grep __AVR_ARCH__ gcc-12.2.0-defs/*.pre| awk '{if($3 >= 100) print; }' 
gcc-12.2.0-defs/atmega1608.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega1609.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega3208.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega3209.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega4808.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega4809.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega808.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/atmega809.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny10.pre:#define __AVR_ARCH__ 100
gcc-12.2.0-defs/attiny1604.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny1606.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny1607.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny1614.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny1616.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny1617.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny202.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny204.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny20.pre:#define __AVR_ARCH__ 100
gcc-12.2.0-defs/attiny212.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny214.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny3214.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny3216.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny3217.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny402.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny404.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny406.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny40.pre:#define __AVR_ARCH__ 100
gcc-12.2.0-defs/attiny412.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny414.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny416.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny417.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny4.pre:#define __AVR_ARCH__ 100
gcc-12.2.0-defs/attiny5.pre:#define __AVR_ARCH__ 100
gcc-12.2.0-defs/attiny804.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny806.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny807.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny814.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny816.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny817.pre:#define __AVR_ARCH__ 103
gcc-12.2.0-defs/attiny9.pre:#define __AVR_ARCH__ 100
gcc-12.2.0-defs/atxmega128a1.pre:#define __AVR_ARCH__ 107
gcc-12.2.0-defs/atxmega128a1u.pre:#define __AVR_ARCH__ 107
gcc-12.2.0-defs/atxmega128a3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega128a3u.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega128a4u.pre:#define __AVR_ARCH__ 107
gcc-12.2.0-defs/atxmega128b1.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega128b3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega128c3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega128d3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega128d4.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega16a4.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega16a4u.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega16c4.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega16d4.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega16e5.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega192a3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega192a3u.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega192c3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega192d3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega256a3b.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega256a3bu.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega256a3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega256a3u.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega256c3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega256d3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega32a4.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega32a4u.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega32c3.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega32c4.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega32d3.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega32d4.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega32e5.pre:#define __AVR_ARCH__ 102
gcc-12.2.0-defs/atxmega384c3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega384d3.pre:#define __AVR_ARCH__ 106
gcc-12.2.0-defs/atxmega64a1.pre:#define __AVR_ARCH__ 105
gcc-12.2.0-defs/atxmega64a1u.pre:#define __AVR_ARCH__ 105
gcc-12.2.0-defs/atxmega64a3.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64a3u.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64a4u.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64b1.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64b3.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64c3.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64d3.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega64d4.pre:#define __AVR_ARCH__ 104
gcc-12.2.0-defs/atxmega8e5.pre:#define __AVR_ARCH__ 102
```

 - 100 seems to be the vvv old `TPI` i/f ones
 - 103 the (non-Dx?) parts with `UDPI` i/f
 - 102 104 105 106 107 are all ATxmega (`PDI` i/f)
 
Didn't spot any 101.

SpenceKonde, 2023-01-23:
You're using old atpacks. 

The TPI interface ones (the AVRrc devices - they've also got only r16-r31, and the instruction timing is different, and direct addressed load/store is a 16-bit opcode with 128-bit address space because the memory on those things is fucking tiny) are not particularly old either - their specs are crap because they're sold for the price of dirt and have to fit in a tiny package.

The 102, 104, arch codes were used for the xmega parts too, yeah. As always, with Atmel running the show, they weren't as consistent. Technically they don't set the size of the flash, but rather are associated with maximums. 

For all Dx and Ex with 32k flash or less, it's 103, for all Dx and Ex with 64k it's 102, and for all Dx with 128k it's 104. A future 256k part would probably be 106. I don't think 101 ever saw the light of day. 105 and 107 I think are related to those parts supporting large external memories.

mcuee, 2023-01-24:
@stefanrueger 

One easy way to get a working compiler is to get the one from @SpenceKonde.

For Linux x86_64
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2

Ref:
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/bootloaders/optiboot_dx/README.md