stefanrueger/urboot #6: ATtiny2313 issues

MCUdude, 2022-11-18:
I just stumbled across this while looking into #5.

I've successfully flashed a vector bootloader, but for some reason, I'm getting a verification mismatch when flashing the user program. It's just a simple blink program compiled using ATTinyCore. What is happening, and why am I getting a verification mismatch?

```
$ ./avrdude -cusbasp -pattiny2313 -Uflash:w:/Users/hans/Downloads/urboot/bootloaders/attiny2313/autobaud/attiny2313_autobaud_ee_lednop_fr_ce_ur_vbl.hex:i 

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e910a (probably t2313)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file /Users/hans/Downloads/urboot/bootloaders/attiny2313/autobaud/attiny2313_autobaud_ee_lednop_fr_ce_ur_vbl.hex for flash
         with 346 bytes in 2 sections within [0x6a0, 0x7ff]
         using 11 pages and 6 pad bytes
avrdude: writing 346 bytes flash ...

Writing | ################################################## | 100% 0.00 s 

avrdude: 346 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/urboot/bootloaders/attiny2313/autobaud/attiny2313_autobaud_ee_lednop_fr_ce_ur_vbl.hex

Reading | ################################################## | 100% 0.00 s 

avrdude: 346 bytes of flash verified

avrdude done.  Thank you.

$ ./avrdude -curclock -pattiny2313 -P /dev/cu.SLAB_USBtoUART -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e910a (probably t2313)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex for flash
         with 456 bytes in 1 section within [0, 0x1c7]
         using 15 pages and 24 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 456 bytes flash ...

Writing | ################################################## | 100% 0.06 s 

avrdude: 456 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex

Reading |                                                    | 0% 0.00 s avrdude: en passant forcing reset vector to point to vector bootloader
Reading | ################################################## | 100% 0.06 s 

avrdude error: verification mismatch, first encountered at addr 0x0002
        device 0xff != input 0x21
avrdude error: verification mismatch

avrdude done.  Thank you.

$ cat /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex
:1000000014C021C020C01FC01EC01DC07FC01BC0A7
:100010001AC019C018C017C016C015C014C013C02C
:1000200012C011C010C00FC00EC011241FBECFEDF2
:10003000CDBF20E0A0E6B0E001C01D92A936B20716
:10004000E1F7AED0BFC0DCCF3FB7F8948091650038
:1000500090916600A0916700B091680022B708B641
:1000600001FE05C02F3F19F00196A11DB11D3FBF34
:10007000BA2FA92F982F8827BC01CD01620F711DBF
:10008000811D911D42E0660F771F881F991F4A95B9
:10009000D1F70895CF92DF92EF92FF92CF93DF9343
:1000A000D3DFEB0188EEC82E83E0D82EE12CF12CB3
:1000B000CBDF6C1B7D0B683E7340A0F0C114D104F4
:1000C000E104F10439F4DF91CF91FF90EF90DF90DC
:1000D000CF90089581E0C81AD108E108F108C8510D
:1000E000DC4FE6CFC114D104E104F10409F7EBCFF2
:1000F00090B79F7D90BF811105C08FB7F8949598F8
:100100008FBF08958FB7F894959AFACF1F920F92E8
:100110000FB60F9211242F933F938F939F93AF931A
:10012000BF938091610090916200A0916300B091B3
:1001300064003091600023E0230F2D3768F126E83A
:10014000230F0296A11DB11D2093600080936100D2
:1001500090936200A0936300B09364008091650067
:1001600090916600A0916700B09168000196A11D72
:10017000B11D8093650090936600A0936700B093D3
:100180006800BF91AF919F918F913F912F910F90F8
:100190000FBE0F901F9018950196A11DB11DD4CFD1
:1001A00083E080BF83BF789489B7826089BF8FB7AF
:1001B000F8948D9A8FBF81E09BDF6CDF80E098DF41
:0801C00069DFF9CFF894FFCFCD
:00000001FF
```

MCUdude, 2022-11-18:
Sorry, false alarm! It was the efuse `selfprogen` setting! For some reason I forgot to program this fuse.

```
$ ./avrdude -cusbasp -pattiny2313 -Ulfuse:r:-:h -Uhfuse:r:-:h -Uefuse:r:-:h

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e910a (probably t2313)
avrdude: reading lfuse memory ...
avrdude: writing output file <stdout>
0xff
avrdude: reading hfuse memory ...
avrdude: writing output file <stdout>
0x9f
avrdude: reading efuse memory ...
avrdude: writing output file <stdout>
0xff

avrdude done.  Thank you.
```

```
$ ./avrdude -curclock -pattiny2313 -P /dev/cu.SLAB_USBtoUART -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e910a (probably t2313)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex for flash
         with 456 bytes in 1 section within [0, 0x1c7]
         using 15 pages and 24 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 456 bytes flash ...

Writing | ################################################## | 100% 0.19 s 

avrdude: 456 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_534228/Blink.ino.hex

Reading | ################################################## | 100% 0.06 s 

avrdude: 456 bytes of flash verified

avrdude done.  Thank you.
```

stefanrueger, 2022-11-18:
> It was the efuse selfprogen setting! 

Phew! I started worrying :) Sounds like we need to be much more detailed in our advice how to put bootloaders on. I already have expanded the [readme](https://github.com/stefanrueger/urboot/blob/main/README.md) on this, so may need to expand. Thanks for mentioning all these little things. Worth documenting how to use/install the bootloaders.

mcuee, 2022-11-19:
Yes, fuse settings can be tricky. So it will be good to show some fuse settings visually. 

As mentioned in other place, I use AVR Fuse Calculator quite a bit and I use the AVRDUESS GUI occassionally just because of fuse settings.