stefanrueger/urboot #7: Enable alternative bootloader entry method

mcuee, 2022-11-19:
I am not so sure if this is really in scope for urboot or not/

The main problem I have is that many of my boards are not meant to be Arduino compatible -- they are simple breakout board and some of them have USB to TTL converter on board. Some of them have RESET button. But many of them do not have the DTR RC reset circuitry.
Ref: the typical minimum setup from MiniCore.
https://github.com/MCUdude/MiniCore#minimal-setup (10k ohm resistor and 100nF capacitor)

Yes I can use another USB to TTL converter and manually add the RC circuitry. But it can be a hassle.

The solution is to use an alternative entry method, typically used by many bootloaders -- using a GPIO pin and pull-up/pull-down to enter the bootloader, example is like AVR109 which detects Port Pin low status to enter bootloader.
https://ww1.microchip.com/downloads/en/Appnotes/doc1644.pdf

Same for AVR109 compatible xboot which has the option USE_ENTER_PIN. It has other options like USE_ENTER_DELAY and USE_ENTER_UART.
https://github.com/alexforencich/xboot#33-bootloader-entrance-options

stefanrueger, 2022-11-19:
Using a separate port pin as bootloader entry condition has disadvantages:
  - Makes the bootloader bigger
  - Requires pullup/down on one GPIO and a button for that purpose on the board
  - This GPIO might not be available for some projects
  - It further explodes the number of possible bootloaders by the extra dimension of entry pin
 
I suggest for *manual reset on a board without reset-circuit* upload by
 1. Reset the board manually; then (within the WDT timeout)
 2. Start avrdude
 
For automated reset with a reset circuit on board it could be that the board takes a loooong time to get out of reset, by which time avrdude had already sent the GET_SYNC; investigate the `SUT_CKSEL` values in the fuses, or check whether there are particularly large caps in the reset circuit. In this case `-xdelay=50` might be needed in the avrdude -curclock call.

mcuee, 2022-11-20:
Fair enough. I will close this one.

closed 2022-11-20 by mcuee