stefanrueger/urboot #8: Better documentation of bootloader firrmware building process

mcuee, 2022-11-21:
It is good to have some documentation for the users to build their own firmware.

The comments here are a good start.
* https://github.com/stefanrueger/urboot/issues/5#issuecomment-1320894654

1. Using Make
```
make MCU=atmega328p F_CPU=16000000L WDTO=1S AUTOBAUD=1 LED=AtmelPB5 DUAL=0 EEPROM=1 URPROTOCOL=1 FRILLS=7 MOVETO=atmega328p_autobaud_ee_led+b5_fr_ce_ur_vbl
```

2. Using avr-gcc directly
```
./avr-toolchain/5.4.0/bin/avr-gcc -DSTART=0x7e80UL -DRJMPWP=0xcfd6 -Wl,--section-start=.text=0x7e80 -Wl,--section-start=.version=0x7ffa -D_urboot_AVAILABLE=22 -g -Wundef -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L -Wno-clobbered -DWDTO=1S -DAUTOBAUD=1 -DLED=AtmelPB5 -DDUAL=0 -DEEPROM=1 -DURPROTOCOL=1 -DFRILLS=7 -DVBL=1 -Wl,--relax -nostartfiles -nostdlib -o atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.elf urboot.c

./avr-toolchain/5.4.0/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.elf atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.hex

./avr-toolchain/5.4.0/bin/avr-objdump -h -S atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.elf > atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.lst

mv atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.hex atmega328p_autobaud_ee_led+b5_fr_ce_ur_vbl.hex
-- 376 384 u7.7 weu-jPrac atmega328p_autobaud_ee_led+b5_fr_ce_ur_vbl.hex
```

mcuee, 2022-11-21:
A few things are nice to have as well.

1) Better documentation of `urboot-gcc` and `hexls`. For example, how to use different toolchaine with `urboot-gcc`, say using another installation of `avr-gcc`.

2) It is also good to document how to make `urboot-gcc` and `hexls` working by mentioning the perl module used.

3) How to modify the Makefile to use another gcc toolchain (still using urboot-gcc wrapper)?

4) How to modify Makefile for users who can not use urboot-gcc for some reason, just plain avr-gcc.

stefanrueger, 2022-11-21:
> 2. It is also good to document how to make urboot-gcc and hexls working by mentioning the perl module used.

Can you help? Being a seasoned perl programmer, I do not know what hurdles there are for someone without perl experience or a perl installation. You managed. What did you have to do? @mcuee 

> 4. How to modify Makefile for users who can not use urboot-gcc for some reason, just plain avr-gcc.

Absolutely not recommended. I only mentioned to you as a highly competent engineer and only for a quick fix to get you to compile one particular bootloader. In reality there are so many pitfalls when using this route, eg, how to compute START and JMPWP, both of which `urboot-gcc` does.

mcuee, 2022-11-21:
I see. So the recommendation is to use `urboot-gcc` to work. I think I can help to mention the perl module which needs to be installed, even though I do not much about perl itself.

Windows is probably the more challenging part if I want to use MSYS2 as it lacks the necessary FILE::HomeDir module. I do not have issues with Linux and macOS.

stefanrueger, 2022-11-21:
> Windows

Only useful if there is the avr-gcc toolchain. urboot.c only compiles with that specific compiler. Best experience is with 4.8.1 and 5.4.0,

mcuee, 2022-11-21:
> > Windows
> 
> Only useful if there is the avr-gcc toolchain. urboot.c only compiles with that specific compiler. Best experience is with 4.8.1 and 5.4.0,

Yes MSYS2 has avr-gcc toolchain. 

I can also try out Atmel provided avr-gcc toolchain with standalone perl installation without MSYS2 (but I tend to use MSYS2 a lot under Windows).

mcuee, 2022-11-21:
Here is an example of using avr-gcc from MSYS2.
```
MINGW64 /c/work/avr/urboot/src
$ avr-gcc -DSTART=0x7e80UL -DRJMPWP=0xcfd6 -Wl,--section-start=.text=0x7e80 -Wl,--section-start=.version=0x7ffa -D_urboot_AVAILABLE=22 -g -Wundef -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L -Wno-clobbered -DWDTO=1S -DAUTOBAUD=1 -DLED=AtmelPB5 -DDUAL=0 -DEEPROM=1 -DURPROTOCOL=1 -DFRILLS=7 -DVBL=1 -Wl,--relax -nostartfiles -nostdlib -o atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.elf urboot.c
urboot.c:1327:19: warning: call-clobbered register used for global register variable
 register uint16_t zaddress asm("r30");
                   ^~~~~~~~
urboot.c:1327:19: warning: call-clobbered register used for global register variable

$ avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.elf atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.hex

$ avr-objdump -h -S atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.elf > atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.lst

$ mv atmega328p_16mhz_auto_115200bps_ev1f7_ledPB5_wdto1s_ur.hex atmega328p_autobaud_ee_led+b5_fr_ce_ur_vbl_wdto1s.hex

```

mcuee, 2022-11-21:
For `hexls`, [Strawberry Perl ](https://strawberryperl.com/)works out of the box.

```
c:\work\avr\urboot\src>perl hexls
376 384 u7.7 weu-jPrac atmega328p_autobaud_ee_led+b5_fr_ce_ur_vbl_wdto1s.hex
376 384 u7.7 weu-jPrac atmega328p_autobaud_ee_led+b5_fr_ce_ur_vbl_wdto8s.hex

```

stefanrueger, 2022-11-21:
> So it seems to me the newer toolchain is pretty much the same in terms of code generration (code size).

One file is not enough to judge. Try
```
(make clean; make all) >& /dev/null
hexls *.hex >/tmp/a
(make clean; make all TOOLVER=8.5.0) >& /dev/null
hexls *.hex | diff /tmp/a -
make clean >& /dev/null
rm /tmp/a
```
This shows you where the two toolchains differ in generated code size

mcuee, 2022-11-21:
> One file is not enough to judge. Try
>
> ```
> (make clean; make all) >& /dev/null
> hexls *.hex >/tmp/a
> (make clean; make all TOOLVER=8.5.0) >& /dev/null
> hexls *.hex | diff /tmp/a -
> make clean >& [...]

Here is the comparison with Atmel provided Linux toolchain.
```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ ./hexls *.hex | diff /tmp/a -
2c2
< 446 512 u7.7 wes-hpr-c anarduino.hex
---
> 462 512 u7.7 wes-hpr-c anarduino.hex
7c7
< 504 512 u7.7 wes-vpr-- atmega1284p_v2.hex
---
> 522 768 u7.7 wes-vpr-- atmega1284p_v2.hex
12c12
< 634 768 u7.7 wes-Vpr-c atmega2560_v3.hex
---
> 660 768 u7.7 wes-Vpr-c atmega2560_v3.hex
21,22c21,22
< 446 512 u7.7 wes-hpr-c atmega328p.hex
< 468 512 u7.7 wes-hpr-c atmega328p_led9_50Hz_fp9.hex
---
> 462 512 u7.7 wes-hpr-c atmega328p.hex
> 484 512 u7.7 wes-hpr-c atmega328p_led9_50Hz_fp9.hex
26,27c26,27
< 436 512 u7.7 wes-hpr-c atmega32.hex
< 244 256 u7.7 w-u-jpr-- atmega32_min.hex
---
> 460 512 u7.7 wes-hpr-c atmega32.hex
> 254 256 u7.7 w-u-jpr-- atmega32_min.hex
30c30
< 446 512 u7.7 wes-hpr-c atmega88.hex
---
> 462 512 u7.7 wes-hpr-c atmega88.hex
33,34c33,34
< 434 512 u7.7 wes-hpr-c atmega8.hex
< 244 256 u7.7 w-u-hpr-- atmega8_min.hex
---
> 458 512 u7.7 wes-hpr-c atmega8.hex
> 254 256 u7.7 w-u-hpr-- atmega8_min.hex
39,41c39,41
< 224 224 u7.7 w-u-jpr-- attiny2313_min.hex
< 256 256 u7.7 w-u-jpr-- attiny84_min.hex
< 256 256 u7.7 w-u-jpr-- attiny85_min.hex
---
> 230 256 u7.7 w-u-jpr-- attiny2313_min.hex
> 270 320 u7.7 w-u-jpr-- attiny84_min.hex
> 270 320 u7.7 w-u-jpr-- attiny85_min.hex
43c43
< 446 512 u7.7 wes-hpr-c diecimila.hex
---
> 462 512 u7.7 wes-hpr-c diecimila.hex
46c46
< 446 512 u7.7 wes-hpr-c jeenode.hex
---
> 462 512 u7.7 wes-hpr-c jeenode.hex
50,51c50,51
< 446 512 u7.7 wes-hpr-c moteino.hex
< 446 512 u7.7 wes-hpr-c pro_16mhz.hex
---
> 462 512 u7.7 wes-hpr-c moteino.hex
> 462 512 u7.7 wes-hpr-c pro_16mhz.hex
53c53
< 442 512 u7.7 wes-hpr-c pro_20mhz.hex
---
> 458 512 u7.7 wes-hpr-c pro_20mhz.hex
55c55
< 464 512 u7.7 wes-hpr-c pro_8mhz.hex
---
> 480 512 u7.7 wes-hpr-c pro_8mhz.hex
57,58c57,58
< 446 512 u7.7 wes-hpr-c promini_led13.hex
< 446 512 u7.7 wes-hpr-c promini_led9.hex
---
> 462 512 u7.7 wes-hpr-c promini_led13.hex
> 462 512 u7.7 wes-hpr-c promini_led9.hex
60c60
< 446 512 u7.7 wes-hpr-c timeduino.hex
---
> 462 512 u7.7 wes-hpr-c timeduino.hex
62c62
< 446 512 u7.7 wes-hpr-c urclock.hex
---
> 462 512 u7.7 wes-hpr-c urclock.hex
```

mcuee, 2022-11-21:
Easier to read version.
```
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ ./hexls ./urbootbin/*.hex
494 512 u7.7 weudhpr-c ./urbootbin/anarduino_cs5_dur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/anarduino.hex
496 512 u7.7 weudjpr-- ./urbootbin/atmega1280_dur.hex
248 256 u7.7 w-u-jpr-- ./urbootbin/atmega1280_min.hex
368 512 u7.7 weu-jpr-c ./urbootbin/atmega1280_ur.hex
248 256 u7.7 w-u-jpr-- ./urbootbin/atmega1284p_min.hex
504 512 u7.7 wes-vpr-- ./urbootbin/atmega1284p_v2.hex
252 256 u7.7 w-u-hpr-- ./urbootbin/atmega168_min.hex
496 512 u7.7 weudjpr-- ./urbootbin/atmega2560_dur.hex
248 256 u7.7 w-u-jpr-- ./urbootbin/atmega2560_min.hex
368 512 u7.7 weu-jpr-c ./urbootbin/atmega2560_ur.hex
634 768 u7.7 wes-Vpr-c ./urbootbin/atmega2560_v3.hex
366 384 u7.7 weu-jpr-c ./urbootbin/atmega328p_7875khz_swio_ur.hex
364 384 u7.7 weu-jpr-c ./urbootbin/atmega328p_8000khz_swio_ur.hex
362 384 u7.7 weu-jpr-c ./urbootbin/atmega328p_8125khz_swio_ur.hex
364 384 u7.7 weu-jpr-c ./urbootbin/atmega328p_8mhz_ur.hex
510 512 u7.7 weudhprac ./urbootbin/atmega328p_adur.hex
254 256 u7.7 w-u-jpra- ./urbootbin/atmega328p_amin.hex
362 384 u7.7 weu-jprac ./urbootbin/atmega328p_autobaud_ur.hex
494 512 u7.7 weudhpr-c ./urbootbin/atmega328p_dur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/atmega328p.hex
468 512 u7.7 wes-hpr-c ./urbootbin/atmega328p_led9_50Hz_fp9.hex
252 256 u7.7 w-u-jpr-- ./urbootbin/atmega328p_min.hex
346 384 u7.7 weu-jpr-c ./urbootbin/atmega328p_ur.hex
362 384 u7.7 weu-jpr-c ./urbootbin/atmega328p_ur_testing.hex
436 512 u7.7 wes-hpr-c ./urbootbin/atmega32.hex
244 256 u7.7 w-u-jpr-- ./urbootbin/atmega32_min.hex
342 384 u7.7 weu-jpr-c ./urbootbin/atmega32_ur.hex
250 256 u7.7 w-u-jpr-- ./urbootbin/atmega644p_min.hex
446 512 u7.7 wes-hpr-c ./urbootbin/atmega88.hex
252 256 u7.7 w-u-jpr-- ./urbootbin/atmega88_min.hex
310 320 u7.7 weu-jpr-- ./urbootbin/atmega88_ur.hex
434 512 u7.7 wes-hpr-c ./urbootbin/atmega8.hex
244 256 u7.7 w-u-hpr-- ./urbootbin/atmega8_min.hex
314 320 u7.7 weu-jpr-- ./urbootbin/atmega8_ur.hex
502 512 u7.7 weudjpr-c ./urbootbin/attiny167_dur.hex
252 256 u7.7 w-u-jpr-- ./urbootbin/attiny167_min.hex
360 384 u7.7 weu-jpr-c ./urbootbin/attiny167_ur.hex
224 224 u7.7 w-u-jpr-- ./urbootbin/attiny2313_min.hex
256 256 u7.7 w-u-jpr-- ./urbootbin/attiny84_min.hex
256 256 u7.7 w-u-jpr-- ./urbootbin/attiny85_min.hex
364 384 u7.7 weu-jpr-c ./urbootbin/attiny85_ur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/diecimila.hex
364 384 u7.7 weu-jpr-c ./urbootbin/digispark.hex
360 384 u7.7 weu-jpr-c ./urbootbin/digisparkpro.hex
446 512 u7.7 wes-hpr-c ./urbootbin/jeenode.hex
364 384 u7.7 weu-jpr-c ./urbootbin/lilypad_ur.hex
374 384 u7.7 weu-jpr-c ./urbootbin/luminet_baud9600_ur.hex
486 512 u7.7 weudhpr-c ./urbootbin/moteino_cs8_dur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/moteino.hex
446 512 u7.7 wes-hpr-c ./urbootbin/pro_16mhz.hex
346 384 u7.7 weu-jpr-c ./urbootbin/pro_16mhz_ur.hex
442 512 u7.7 wes-hpr-c ./urbootbin/pro_20mhz.hex
342 384 u7.7 weu-jpr-c ./urbootbin/pro_20mhz_ur.hex
464 512 u7.7 wes-hpr-c ./urbootbin/pro_8mhz.hex
364 384 u7.7 weu-jpr-c ./urbootbin/pro_8mhz_ur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/promini_led13.hex
446 512 u7.7 wes-hpr-c ./urbootbin/promini_led9.hex
486 512 u7.7 weudhpr-c ./urbootbin/timeduino_cs8_dur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/timeduino.hex
486 512 u7.7 weudhpr-c ./urbootbin/urclock_cs8_dur.hex
446 512 u7.7 wes-hpr-c ./urbootbin/urclock.hex
mcuee@UbuntuSwift3:~/build/avr/urboot/src$ ./hexls ./urbootbin_gcc730/*.hex
494 512 u7.7 weudhpr-c ./urbootbin_gcc730/anarduino_cs5_dur.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/anarduino.hex
496 512 u7.7 weudjpr-- ./urbootbin_gcc730/atmega1280_dur.hex
248 256 u7.7 w-u-jpr-- ./urbootbin_gcc730/atmega1280_min.hex
[... 35 lines cut ...]
270 320 u7.7 w-u-jpr-- ./urbootbin_gcc730/attiny84_min.hex
270 320 u7.7 w-u-jpr-- ./urbootbin_gcc730/attiny85_min.hex
364 384 u7.7 weu-jpr-c ./urbootbin_gcc730/attiny85_ur.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/diecimila.hex
364 384 u7.7 weu-jpr-c ./urbootbin_gcc730/digispark.hex
360 384 u7.7 weu-jpr-c ./urbootbin_gcc730/digisparkpro.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/jeenode.hex
364 384 u7.7 weu-jpr-c ./urbootbin_gcc730/lilypad_ur.hex
374 384 u7.7 weu-jpr-c ./urbootbin_gcc730/luminet_baud9600_ur.hex
486 512 u7.7 weudhpr-c ./urbootbin_gcc730/moteino_cs8_dur.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/moteino.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/pro_16mhz.hex
346 384 u7.7 weu-jpr-c ./urbootbin_gcc730/pro_16mhz_ur.hex
458 512 u7.7 wes-hpr-c ./urbootbin_gcc730/pro_20mhz.hex
342 384 u7.7 weu-jpr-c ./urbootbin_gcc730/pro_20mhz_ur.hex
480 512 u7.7 wes-hpr-c ./urbootbin_gcc730/pro_8mhz.hex
364 384 u7.7 weu-jpr-c ./urbootbin_gcc730/pro_8mhz_ur.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/promini_led13.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/promini_led9.hex
486 512 u7.7 weudhpr-c ./urbootbin_gcc730/timeduino_cs8_dur.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/timeduino.hex
486 512 u7.7 weudhpr-c ./urbootbin_gcc730/urclock_cs8_dur.hex
462 512 u7.7 wes-hpr-c ./urbootbin_gcc730/urclock.hex

```

mcuee, 2022-11-22:
> > I can help to mention the perl module which needs to be installed
> 
> Please do

The following perl modules are required by `hexls` and `urboot-gcc`.
```
File::Basename;
File::HomeDir;
Capture::Tiny ':all'
```

Ubuntu 20.04 -- hexls and urboot-gcc work out of the box as perl and necessary perl-modules have been installed. I do not see File::HomeDir listed but it still works. 
https://packages.ubuntu.com/focal/all/perl-modules-5.30/filelist

macOS: install File::HomeDir using sudo if you are using system provided perl. It should also work if you use other perl version.

Windows: install [Strawberry Perl ](https://strawberryperl.com/) and you are all set. You can also use other Perl installations.

mcuee, 2022-11-22:
BTW, I am not able to get File::HomeDir installed under MSYS2.

```
$ ./hexls
Can't locate File/HomeDir.pm in @INC (you may need to install the File::HomeDir module) (@INC contains: /usr/lib/perl5/site_perl /usr/share/perl5/site_perl /usr/lib/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib/perl5/core_perl /usr/share/perl5/core_perl) at ./hexls line 16.
BEGIN failed--compilation aborted at ./hexls line 16.

$ cpan install File::HomeDir
Loading internal logger. Log::Log4perl recommended for better logging
Fetching with LWP:
http://www.cpan.org/authors/01mailrc.txt.gz
Reading '/c/Users/xfchen/.cpan/sources/authors/01mailrc.txt.gz'
............................................................................DONE
Fetching with LWP:
http://www.cpan.org/modules/02packages.details.txt.gz
Reading '/c/Users/xfchen/.cpan/sources/modules/02packages.details.txt.gz'
  Database was generated on Tue, 22 Nov 2022 04:53:55 GMT
..............
  New CPAN.pm version (v2.34) available.
  [Currently running version is v2.27]
  You might want to try
    install CPAN
    reload cpan
  to both upgrade CPAN.pm and run the new version without leaving
  the current session.

..............................................................DONE
Fetching with LWP:
http://www.cpan.org/modules/03modlist.data.gz
Reading '/c/Users/xfchen/.cpan/sources/modules/03modlist.data.gz'
DONE
Writing /c/Users/xfchen/.cpan/Metadata
Running install for module 'File::HomeDir'
Fetching with LWP:
http://www.cpan.org/authors/id/R/RE/REHSACK/File-HomeDir-1.006.tar.gz
Fetching with LWP:
http://www.cpan.org/authors/id/R/RE/REHSACK/CHECKSUMS
Checksum for /c/Users/xfchen/.cpan/sources/authors/id/R/RE/REHSACK/File-HomeDir-1.006.tar.gz ok
'YAML' not installed, will not store persistent state
Configuring R/RE/REHSACK/File-HomeDir-1.006.tar.gz with Makefile.PL
Checking if your kit is complete...
Looks good
Warning: prerequisite File::Which 0.05 not found.
Generating a Unix-style Makefile
Writing Makefile for File::HomeDir
Writing MYMETA.yml and MYMETA.json
  REHSACK/File-HomeDir-1.006.tar.gz
  /usr/bin/perl Makefile.PL INSTALLDIRS=site -- OK
Running make for R/RE/REHSACK/File-HomeDir-1.006.tar.gz
---- Unsatisfied dependencies detected during ----
----     REHSACK/File-HomeDir-1.006.tar.gz    ----
    File::Which [requires]
Running install for module 'File::Which'
Fetching with LWP:
http://www.cpan.org/authors/id/P/PL/PLICEASE/File-Which-1.27.tar.gz
Fetching with LWP:
http://www.cpan.org/authors/id/P/PL/PLICEASE/CHECKSUMS
Checksum for /c/Users/xfchen/.cpan/sources/authors/id/P/PL/PLICEASE/File-Which-1.27.tar.gz ok
Configuring P/PL/PLICEASE/File-Which-1.27.tar.gz with Makefile.PL
Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for File::Which
Writing MYMETA.yml and MYMETA.json
  PLICEASE/File-Which-1.27.tar.gz
  /usr/bin/perl Makefile.PL INSTALLDIRS=site -- OK
Running make for P/PL/PLICEASE/File-Which-1.27.tar.gz
cp lib/File/Which.pm blib/lib/File/Which.pm
Manifying 1 pod document
  PLICEASE/File-Which-1.27.tar.gz
  /usr/bin/make -- OK
The current configuration of allow_installing_outdated_dists is 'ask/no', but for this option we would need 'CPAN::DistnameInfo' installed. Please install 'CPAN::DistnameInfo' as soon as possible. As long as we are not equipped with 'CPAN::DistnameInfo' this option does not take effect
Running make test for PLICEASE/File-Which-1.27.tar.gz
  PLICEASE/File-Which-1.27.tar.gz
  /usr/bin/make test -- NOT OK
//hint// to see the cpan-testers results for installing this module, try:
  reports PLICEASE/File-Which-1.27.tar.gz
  REHSACK/File-HomeDir-1.006.tar.gz
  Has already been unwrapped into directory /c/Users/xfchen/.cpan/build/File-HomeDir-1.006-0
  REHSACK/File-HomeDir-1.006.tar.gz
  Has already been prepared
Running make for R/RE/REHSACK/File-HomeDir-1.006.tar.gz
Warning: Prerequisite 'File::Which => 0.05' for 'REHSACK/File-HomeDir-1.006.tar.gz' failed when processing 'PLICEASE/File-Which-1.27.tar.gz' with 'make_test => NO'. Continuing, but chances to succeed are limited.
cp lib/File/HomeDir/Test.pm blib/lib/File/HomeDir/Test.pm
cp lib/File/HomeDir/Driver.pm blib/lib/File/HomeDir/Driver.pm
cp lib/File/HomeDir/Darwin/Cocoa.pm blib/lib/File/HomeDir/Darwin/Cocoa.pm
cp lib/File/HomeDir/Windows.pm blib/lib/File/HomeDir/Windows.pm
cp lib/File/HomeDir/Darwin.pm blib/lib/File/HomeDir/Darwin.pm
cp lib/File/HomeDir/Unix.pm blib/lib/File/HomeDir/Unix.pm
cp lib/File/HomeDir/Darwin/Carbon.pm blib/lib/File/HomeDir/Darwin/Carbon.pm
cp lib/File/HomeDir.pm blib/lib/File/HomeDir.pm
cp lib/File/HomeDir/FreeDesktop.pm blib/lib/File/HomeDir/FreeDesktop.pm
cp lib/File/HomeDir/MacOS9.pm blib/lib/File/HomeDir/MacOS9.pm
Manifying 10 pod documents
  REHSACK/File-HomeDir-1.006.tar.gz
  /usr/bin/make -- OK
The current configuration of allow_installing_outdated_dists is 'ask/no', but for this option we would need 'CPAN::DistnameInfo' installed. Please install 'CPAN::DistnameInfo' as soon as possible. As long as we are not equipped with 'CPAN::DistnameInfo' this option does not take effect
Running make test for REHSACK/File-HomeDir-1.006.tar.gz
  REHSACK/File-HomeDir-1.006.tar.gz
one dependency not OK (File::Which); additionally test harness failed
  /usr/bin/make test -- NOT OK
//hint// to see the cpan-testers results for installing this module, try:
  reports REHSACK/File-HomeDir-1.006.tar.gz

```

mcuee, 2022-11-22:
Hack to get things working under Windows with Strawberry Perl and Atmel 7.3.0 toolchain.
```
C:\work\avr\urboot [main ≡ +63 ~2 -0 !]> git diff
diff --git a/src/Makefile b/src/Makefile
index 1a30321dc2..90cb91403e 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -249,7 +249,7 @@ endif
 # UPROTOCOL with EEPROM compiles much neater with 5.4.0
 ifdef URPROTOCOL
 ifeq (1, $(EEPROM))
-override TOOLVER := 5.4.0
+override TOOLVER := 7.3.0
 override GCCROOT = ./avr-toolchain/$(TOOLVER)/bin/
 endif
 endif
@@ -257,7 +257,7 @@ endif
 # Toolchain 4.8.1 does not know some MCUs, bump to 5.4.0
 ifdef MCU
 ifeq (/$(MCU)/, $(findstring /$(MCU)/, /ata5702m322/ata5782/ata5791/ata5831/ata8210/ata8510/atmega168pb/atmega328pb/atmega48pb/atmega88pb/))
-override TOOLVER := 5.4.0
+override TOOLVER := 7.3.0
 override GCCROOT = ./avr-toolchain/$(TOOLVER)/bin/
 endif
 endif
@@ -269,11 +269,11 @@ WHAT := $(WHAT)_v$(VERSION)
 endif

 # Older avr-gcc versions such as 4.8.1 or 5.4.0 produce tighter code
-TOOLVER ?= 4.8.1
+TOOLVER ?= 7.3.0
 GCCROOT ?= ./avr-toolchain/$(TOOLVER)/bin/
 MK_OPTS  = TOOLVER=$(TOOLVER) GCCROOT=$(GCCROOT)

-URBOOTGCC= ./urboot-gcc -toolchain=$(TOOLVER)
+URBOOTGCC= perl ./urboot-gcc -toolchain=$(TOOLVER)
 OBJCOPY        = $(GCCROOT)avr-objcopy
 OBJDUMP        = $(GCCROOT)avr-objdump
 SIZE           = $(GCCROOT)avr-size
@@ -290,7 +290,7 @@ bootloader: $(WHAT).hex

 min: attiny2313_min atmega1280_min atmega1284p_min atmega168_min atmega2560_min atmega32_min atmega328p_min \
   atmega644p_min atmega8_min atmega88_min attiny167_min attiny84_min attiny85_min
-       ./hexls -sort
+       perl hexls -sort

 all: attiny2313_min atmega168_min atmega2560_min atmega32_min \
   atmega644p_min atmega8_min atmega88_min attiny84_min \
@@ -327,7 +327,7 @@ all: attiny2313_min atmega168_min atmega2560_min atmega32_min \
   attiny167_min attiny167_ur attiny167_dur digisparkpro \
   attiny85_min attiny85_ur digispark \
   lilypad_ur luminet_baud9600_ur
-       ./hexls -sort *.hex
+       perl hexls -sort *.hex

 # Examples
 #   make MCU=atmega32 F_CPU=11059200L
@@ -729,9 +729,9 @@ ifdef MOVETO
        mv $@ $(MOVETO).hex
        @mv $*.elf $(MOVETO).elf
        @mv $*.lst $(MOVETO).lst
-       @echo -- `./hexls $(MOVETO).hex`
+       @echo -- `perl hexls $(MOVETO).hex`
 else
-       @echo -- `./hexls $@`
+       @echo -- `perl hexls $@`
 endif
        @echo

diff --git a/src/urboot-gcc b/src/urboot-gcc
index f25af48544..5e221c2057 100755
--- a/src/urboot-gcc
+++ b/src/urboot-gcc
@@ -1,4 +1,4 @@
-#!/usr/bin/perl -CS
+#!/usr/bin/perl

 # urboot-gcc
 #

```

mcuee, 2022-11-22:
I managed to find a few Atmel toolchain for Windows. Here is the comparison.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/xfchen/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/xfchen/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

urboot | Atmel 7.3.0 | Atmel 5.4.0 | Atmel 4.6.2   with exceptions on 5.4.0
-- | -- | -- | --
446 512 u7.7   wes-hpr-c anarduino.hex | 462 512 u7.7   wes-hpr-c anarduino.hex | 446 512 u7.7   wes-hpr-c anarduino.hex | 446 512 u7.7   wes-hpr-c anarduino.hex
494 512 u7.7   weudhpr-c anarduino_cs5_dur.hex | 508 512 u7.7   weudhpr-c anarduino_cs5_dur.hex | 494 512 u7.7   weudhpr-c anarduino_cs5_dur.hex | 494 512 u7.7   weudhpr-c anarduino_cs5_dur.hex
496 512 u7.7   weudjpr-- atmega1280_dur.hex | 510 512 u7.7   weudjpr-- atmega1280_dur.hex | 496 512 u7.7   weudjpr-- atmega1280_dur.hex | 500 512 u7.7   weudjpr-- atmega1280_dur.hex
248 256 u7.7   w-u-jpr-- atmega1280_min.hex | 248 256 u7.7   w-u-jpr-- atmega1280_min.hex | 248 256 u7.7   w-u-jpr-- atmega1280_min.hex | 248 256 u7.7   w-u-jpr-- atmega1280_min.hex
386 512 u7.7   weu-jPr-c atmega1280_ur.hex | 392 512 u7.7   weu-jPr-c atmega1280_ur.hex | 386 512 u7.7   weu-jPr-c atmega1280_ur.hex | 390 512 u7.7   weu-jPr-c atmega1280_ur.hex
248 256 u7.7   w-u-jpr-- atmega1284p_min.hex | 248 256 u7.7   w-u-jpr-- atmega1284p_min.hex | 248 256 u7.7   w-u-jpr-- atmega1284p_min.hex | 248 256 u7.7   w-u-jpr-- atmega1284p_min.hex
504 512 u7.7   wes-vpr-- atmega1284p_v2.hex | 522 768 u7.7   wes-vpr-- atmega1284p_v2.hex | 504 512 u7.7   wes-vpr-- atmega1284p_v2.hex | 506 512 u7.7   wes-vpr-- atmega1284p_v2.hex
252 256 u7.7   w-u-hpr-- atmega168_min.hex | 252 256 u7.7   w-u-hpr-- atmega168_min.hex | 252 256 u7.7   w-u-hpr-- atmega168_min.hex | 252 256 u7.7   w-u-hpr-- atmega168_min.hex
496 512 u7.7   weudjpr-- atmega2560_dur.hex | 510 512 u7.7   weudjpr-- atmega2560_dur.hex | 496 512 u7.7   weudjpr-- atmega2560_dur.hex | 500 512 u7.7   weudjpr-- atmega2560_dur.hex
248 256 u7.7   w-u-jpr-- atmega2560_min.hex | 248 256 u7.7   w-u-jpr-- atmega2560_min.hex | 248 256 u7.7   w-u-jpr-- atmega2560_min.hex | 248 256 u7.7   w-u-jpr-- atmega2560_min.hex
386 512 u7.7   weu-jPr-c atmega2560_ur.hex | 392 512 u7.7   weu-jPr-c atmega2560_ur.hex | 386 512 u7.7   weu-jPr-c atmega2560_ur.hex | 390 512 u7.7   weu-jPr-c atmega2560_ur.hex
642 768 u7.7   wes-Vpr-c atmega2560_v3.hex | 668 768 u7.7   wes-Vpr-c atmega2560_v3.hex | 654 768 u7.7   wes-Vpr-c atmega2560_v3.hex | 642 768 u7.7   wes-Vpr-c atmega2560_v3.hex
436 512 u7.7   wes-hpr-c atmega32.hex | 460 512 u7.7   wes-hpr-c atmega32.hex | 442 512 u7.7   wes-hpr-c atmega32.hex | 438 512 u7.7   wes-hpr-c atmega32.hex
244 256 u7.7   w-u-jpr-- atmega32_min.hex | 254 256 u7.7   w-u-jpr-- atmega32_min.hex | 246 256 u7.7   w-u-jpr-- atmega32_min.hex | 256 256 u7.7   w-u-jPr-- atmega32_min.hex
356 384 u7.7   weu-jPr-c atmega32_ur.hex | 372 384 u7.7   weu-jPr-c atmega32_ur.hex | 356 384 u7.7   weu-jPr-c atmega32_ur.hex | 354 384 u7.7   weu-jPr-c atmega32_ur.hex
446 512 u7.7   wes-hpr-c atmega328p.hex | 462 512 u7.7   wes-hpr-c atmega328p.hex | 446 512 u7.7   wes-hpr-c atmega328p.hex | 446 512 u7.7   wes-hpr-c atmega328p.hex
380 384 u7.7   weu-jPr-c atmega328p_7875khz_swio_ur.hex | 372 384 u7.7   weu-jpr-c atmega328p_7875khz_swio_ur.hex | 380 384 u7.7   weu-jPr-c atmega328p_7875khz_swio_ur.hex | 384 384 u7.7   weu-jPr-c atmega328p_7875khz_swio_ur.hex
378 384 u7.7   weu-jPr-c atmega328p_8000khz_swio_ur.hex | 384 384 u7.7   weu-jPr-c atmega328p_8000khz_swio_ur.hex | 378 384 u7.7   weu-jPr-c atmega328p_8000khz_swio_ur.hex | 382 384 u7.7   weu-jPr-c atmega328p_8000khz_swio_ur.hex
376 384 u7.7   weu-jPr-c atmega328p_8125khz_swio_ur.hex | 382 384 u7.7   weu-jPr-c atmega328p_8125khz_swio_ur.hex | 376 384 u7.7   weu-jPr-c atmega328p_8125khz_swio_ur.hex | 380 384 u7.7   weu-jPr-c atmega328p_8125khz_swio_ur.hex
378 384 u7.7   weu-jPr-c atmega328p_8mhz_ur.hex | 384 384 u7.7   weu-jPr-c atmega328p_8mhz_ur.hex | 378 384 u7.7   weu-jPr-c atmega328p_8mhz_ur.hex | 382 384 u7.7   weu-jPr-c atmega328p_8mhz_ur.hex
510 512 u7.7   weudhprac atmega328p_adur.hex | 524 1024 u7.7   weudhprac atmega328p_adur.hex | 510 512 u7.7   weudhprac atmega328p_adur.hex | 510 512 u7.7   weudhprac atmega328p_adur.hex
254 256 u7.7   w-u-jpra- atmega328p_amin.hex | 254 256 u7.7   w-u-jpra- atmega328p_amin.hex | 254 256 u7.7   w-u-jpra- atmega328p_amin.hex | 254 256 u7.7   w-u-jpra- atmega328p_amin.hex
376 384 u7.7   weu-jPrac atmega328p_autobaud_ur.hex | 382 384 u7.7   weu-jPrac atmega328p_autobaud_ur.hex | 376 384 u7.7   weu-jPrac atmega328p_autobaud_ur.hex | 380 384 u7.7   weu-jPrac atmega328p_autobaud_ur.hex
494 512 u7.7   weudhpr-c atmega328p_dur.hex | 508 512 u7.7   weudhpr-c atmega328p_dur.hex | 494 512 u7.7   weudhpr-c atmega328p_dur.hex | 494 512 u7.7   weudhpr-c atmega328p_dur.hex
468 512 u7.7   wes-hpr-c atmega328p_led9_50Hz_fp9.hex | 484 512 u7.7   wes-hpr-c atmega328p_led9_50Hz_fp9.hex | 468 512 u7.7   wes-hpr-c atmega328p_led9_50Hz_fp9.hex | 468 512 u7.7   wes-hpr-c atmega328p_led9_50Hz_fp9.hex
[... cut ...]
486 512 u7.7   weudhpr-c timeduino_cs8_dur.hex | 500 512 u7.7   weudhpr-c timeduino_cs8_dur.hex | 486 512 u7.7   weudhpr-c timeduino_cs8_dur.hex | 486 512 u7.7   weudhpr-c timeduino_cs8_dur.hex
446 512 u7.7   wes-hpr-c urclock.hex | 462 512 u7.7   wes-hpr-c urclock.hex | 446 512 u7.7   wes-hpr-c urclock.hex | 446 512 u7.7   wes-hpr-c urclock.hex
486 512 u7.7   weudhpr-c urclock_cs8_dur.hex | 500 512 u7.7   weudhpr-c urclock_cs8_dur.hex | 486 512 u7.7   weudhpr-c urclock_cs8_dur.hex | 486 512 u7.7   weudhpr-c urclock_cs8_dur.hex

</body>

</html>

mcuee, 2022-11-22:
Maybe the other MSYS2 installation needs some updates. I tried another Windows machine and it works fine to install File::HomeDir. But then it does not work. Therefore I think I will have to rely on Strawberry Perl together with MSYS2 for now.
```
$ make all TOOLVER=5.4.0_win32
make TOOLVER=5.4.0_win32 GCCROOT=./avr-toolchain/5.4.0_win32/bin/ MCU=attiny2313 URPROTOCOL=1 EEPROM=0 VBL=1 FRILLS=2 VBL_VECT_NUM=EEPROM_READY_vect_num MOVETO=attiny2313_min
make[1]: Entering directory '/c/work/avr/urboot/src'
Can't locate File/HomeDir.pm in @INC (you may need to install the File::HomeDir module) (@INC contains: /usr/lib/perl5/site_perl /usr/share/perl5/site_perl /usr/lib/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib/perl5/core_perl /usr/share/perl5/core_perl) at ./urboot-gcc line 18.
BEGIN failed--compilation aborted at ./urboot-gcc line 18.
make[1]: *** [Makefile:724: attiny2313_16mhz_115200bps_v1f2_vnEEPROM_READY_vect_num_wdto1s_ur.hex] Error 2
make[1]: Leaving directory '/c/work/avr/urboot/src'
make: *** [Makefile:345: attiny2313_min] Error 2

$ cpan install File::HomeDir
Loading internal logger. Log::Log4perl recommended for better logging
Reading 'C:\msys64\home\xiaof\.cpan-w64\Metadata'
  Database was generated on Tue, 22 Nov 2022 09:41:01 GMT
CPAN: Module::CoreList loaded ok (v5.20210123)
File::HomeDir is up to date (1.006).

$ ls -la /usr/lib/perl5/core_perl/File/
total 52
drwxr-xr-x 1 xiaof xiaof     0 Jan 22  2022 .
drwxr-xr-x 1 xiaof xiaof     0 Jan 22  2022 ..
-rwxr-xr-x 1 xiaof xiaof  7993 Jan  1  2022 DosGlob.pm
-rw-r--r-- 1 xiaof xiaof 12712 Jan  1  2022 Glob.pm
drwxr-xr-x 1 xiaof xiaof     0 Jan 22  2022 Spec
-rw-r--r-- 1 xiaof xiaof 10593 Jan  1  2022 Spec.pm

$ ls -la /usr/lib/perl5/core_perl/File/Spec
total 96
drwxr-xr-x 1 xiaof xiaof     0 Jan 22  2022 .
drwxr-xr-x 1 xiaof xiaof     0 Jan 22  2022 ..
-rw-r--r-- 1 xiaof xiaof   982 Jan  1  2022 AmigaOS.pm
-rw-r--r-- 1 xiaof xiaof  3627 Jan  1  2022 Cygwin.pm
-rw-r--r-- 1 xiaof xiaof  1571 Jan  1  2022 Epoc.pm
-rw-r--r-- 1 xiaof xiaof  2348 Jan  1  2022 Functions.pm
-rw-r--r-- 1 xiaof xiaof 22273 Jan  1  2022 Mac.pm
-rw-r--r-- 1 xiaof xiaof  6753 Jan  1  2022 OS2.pm
-rw-r--r-- 1 xiaof xiaof 15885 Jan  1  2022 Unix.pm
-rw-r--r-- 1 xiaof xiaof 16305 Jan  1  2022 VMS.pm
-rw-r--r-- 1 xiaof xiaof 10754 Jan  1  2022 Win32.pm
```

mcuee, 2022-11-22:
Under macOS, other than installing `File::HomeDir`, the other thing is that you need to assemble the working avr-gcc directory. For example, I have to merge avr-gcc and avr-binuntils to have a working 9.4.0 toolchain. Unfortunately Atmel provided avr-gcc macOS x96_64 toolchains (7.3.0 and 5.4.0) do not work because of the developer signature warnings which I am lazy to fix.

Since I have only the 9.4.0 toolchain, minor change to the Makefile is required.
```
mcuee@mcuees-Mac-mini urboot % git diff
diff --git a/src/Makefile b/src/Makefile
index 1a30321dc2..d5542cb697 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -249,7 +249,7 @@ endif
 # UPROTOCOL with EEPROM compiles much neater with 5.4.0
 ifdef URPROTOCOL
 ifeq (1, $(EEPROM))
-override TOOLVER := 5.4.0
+override TOOLVER := 9.4.0
 override GCCROOT = ./avr-toolchain/$(TOOLVER)/bin/
 endif
 endif
@@ -257,7 +257,7 @@ endif
 # Toolchain 4.8.1 does not know some MCUs, bump to 5.4.0
 ifdef MCU
 ifeq (/$(MCU)/, $(findstring /$(MCU)/, /ata5702m322/ata5782/ata5791/ata5831/ata8210/ata8510/atmega168pb/atmega328pb/atmega48pb/atmega88pb/))
-override TOOLVER := 5.4.0
+override TOOLVER := 9.4.0
 override GCCROOT = ./avr-toolchain/$(TOOLVER)/bin/
 endif
 endif
@@ -269,7 +269,7 @@ WHAT := $(WHAT)_v$(VERSION)
 endif
 
 # Older avr-gcc versions such as 4.8.1 or 5.4.0 produce tighter code
-TOOLVER ?= 4.8.1
+TOOLVER ?= 9.4.0
 GCCROOT ?= ./avr-toolchain/$(TOOLVER)/bin/
 MK_OPTS  = TOOLVER=$(TOOLVER) GCCROOT=$(GCCROOT)```

mcuee, 2022-11-23:
@stefanrueger 

It seems to me there is nothing much to improve on the build process under Linux (working out of the box) and macOS. You just need to document the perl modules required. 

For Windows, unfortunately MSYS2 perl packaging is not good. We can forget about Windows support for a while. Windows users can refer to this post to build if they really want to do that (using Strawberry Perl and then modify the few files accordingly).

stefanrueger, 2022-11-23:
@mcuee Many thanks for exploring how to use the project under Windows and MacOS, and for sharing tips for better documentation! Well appreciated.

I have updated `urboot-gcc` and `hexls` so they no longer require `File::Homedir`. Great tip.

The comparison table stock urboot/9.4.0/7.3.0 is great! I suspect you mixed the columns for urboot and 9.4.0, though? I believe the stock `urboot` decision is always the best choice, not?

Note that the `urboot` project does **not** recommend to compile with other avr-toolchains than  4.8.1 or 5.4.0. Generally, the code has lots of assembler bits in there that rely on the compiler API for creating code.  It is a bad idea to use newer compilers, as they may well change that API! So, if this projects is ported to Windows/MacOS, it would need the 4.8.1 and 5.4.0 toolchains there! This is one of the few situations where an old version is actually better.

mcuee, 2022-11-23:
@stefanrueger 

Great, now that `hexls` works out of the box under MSYS2 Windows. For `urboot-gcc`, it still complains about 

```
MINGW64 /c/work/avr/urboot/src
$ ./hexls ./all/atmega328p.hex
446 512 u7.7 wes-hpr-c ./all/atmega328p.hex

$ ./urboot-gcc
Can't locate Capture/Tiny.pm in @INC (you may need to install the Capture::Tiny module) (@INC contains: /usr/lib/perl5/site_perl /usr/share/perl5/site_perl /usr/lib/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib/perl5/core_perl /usr/share/perl5/core_perl) at ./urboot-gcc line 18.
BEGIN failed--compilation aborted at ./urboot-gcc line 18.
```

mcuee, 2022-11-23:
> The comparison table stock urboot/9.4.0/7.3.0 is great! I suspect you mixed the columns for urboot and 9.4.0, though? I believe the stock urboot decision is always the best choice, not?

Yes, you are right, urboot is always the best. Table updated

mcuee, 2022-11-23:
> Note that the urboot project does not recommend to compile with other avr-toolchains than 4.8.1 or 5.4.0. Generally, the code has lots of assembler bits in there that rely on the compiler API for [...]

I can not find version 4.8.1 but Atmel provided 7.3.0 and 5.4.0 for Linux, Windows and macOS. Now I need to figure out how to get Atmel avr-gcc compiler to work under macOS (due to all the developer siging issues).

https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers
Atmel version 3.7.0 --> avr-gcc 7.3.0
Atmel version 3.6.2 --> avr-gcc 5.4.0

mcuee, 2022-11-24:
I think this issue can be closed.

closed 2022-11-24 by mcuee

mcuee, 2022-11-26:
> I will report the macOS 5.4.0 situation later.

This is the current problem with the Atmel toolchain avr-gcc 5.4.0. I will see how to fix it.

```
mcuee@mcuees-Mac-mini src % make all TOOLVER=5.4.0_atmel
/Library/Developer/CommandLineTools/usr/bin/make TOOLVER=5.4.0_atmel GCCROOT=./avr-toolchain/5.4.0_atmel/bin/ MCU=attiny2313 URPROTOCOL=1 EEPROM=0 VBL=1 FRILLS=2 VBL_VECT_NUM=EEPROM_READY_vect_num MOVETO=attiny2313_min
./avr-toolchain/5.4.0_atmel/bin/avr-gcc -DSTART=0x400UL -DRJMPWP=0xcfff -Wl,--section-start=.text=0x400 -Wl,--section-start=.version=0x7fa -g -Wundef -Wall -Os -fno-split-wide-types -mrelax -mmcu=attiny2313 -DF_CPU=16000000L -Wno-clobbered -DWDTO=1S -DDUAL=0 -DEEPROM=0 -DURPROTOCOL=1 -DFRILLS=2 -DVBL=1 -DVBL_VECT_NUM=EEPROM_READY_vect_num -Wl,--relax -nostartfiles -nostdlib -o attiny2313_16mhz_115200bps_v1f2_vnEEPROM_READY_vect_num_wdto1s_ur.elf urboot.c
/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: /Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/bin/../libexec/gcc/avr/5.4.0/liblto_plugin.so: error loading plugin: dlopen(/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/bin/../libexec/gcc/avr/5.4.0/liblto_plugin.so, 0x0002): tried: '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/bin/../libexec/gcc/avr/5.4.0/liblto_plugin.so' (code signature in <66B005B7-D146-3181-8520-2C46E5E47AD6> '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' not valid for use in process: library load disallowed by system policy), '/System/Volumes/Preboot/Cryptexes/OS/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/bin/../libexec/gcc/avr/5.4.0/liblto_plugin.so' (no such file), '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/bin/../libexec/gcc/avr/5.4.0/liblto_plugin.so' (code signature in <66B005B7-D146-3181-8520-2C46E5E47AD6> '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' not valid for use in process: library load disallowed by system policy), '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' (code signature in <66B005B7-D146-3181-8520-2C46E5E47AD6> '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' not valid for use in process: library load disallowed by system policy), '/System/Volumes/Preboot/Cryptexes/OS/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' (no such file), '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' (code signature in <66B005B7-D146-3181-8520-2C46E5E47AD6> '/Users/mcuee/build/avr/urboot/src/avr-toolchain/5.4.0_atmel/libexec/gcc/avr/5.4.0/liblto_plugin.0.so' not valid for use in process: library load disallowed by system policy)
collect2: error: ld returned 1 exit status
./avr-toolchain/5.4.0_atmel/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex attiny2313_16mhz_115200bps_v1f2_vnEEPROM_READY_vect_num_wdto1s_ur.elf attiny2313_16mhz_115200bps_v1f2_vnEEPROM_READY_vect_num_wdto1s_ur.hex
./avr-toolchain/5.4.0_atmel/bin/avr-objcopy: 'attiny2313_16mhz_115200bps_v1f2_vnEEPROM_READY_vect_num_wdto1s_ur.elf': No such file
make[1]: *** [attiny2313_16mhz_115200bps_v1f2_vnEEPROM_READY_vect_num_wdto1s_ur.hex] Error 1
make: *** [attiny2313_min] Error 2
```

mcuee, 2022-11-26:
Work-around is to unlock the file.
https://support.blackfire.io/en/articles/3669492-issues-with-macos-catalina

mcuee, 2022-11-26:
Results from Atmel 5.4.0 toolchain -- it may produce some bigger size urboot hex files but the space occupied is the same in the end.

Attribute | Name | Size (urboot) | Space (urboot) | Size (Atmel 5.4.0) | Space (Atmel 5.4.0)
-- | -- | -- | -- | -- | --
wes-hpr-c | anarduino.hex | 446 | 512 | 446 | 512
weudhpr-c | anarduino_cs5_dur.hex | 494 | 512 | 494 | 512
weudjpr-- | atmega1280_dur.hex | 496 | 512 | 496 | 512
w-u-jpr-- | atmega1280_min.hex | 248 | 256 | 248 | 256
weu-jPr-c | atmega1280_ur.hex | 386 | 512 | 386 | 512
w-u-jpr-- | atmega1284p_min.hex | 248 | 256 | 248 | 256
wes-vpr-- | atmega1284p_v2.hex | 504 | 512 | 504 | 512
w-u-hpr-- | atmega168_min.hex | 252 | 256 | 252 | 256
weudjpr-- | atmega2560_dur.hex | 496 | 512 | 496 | 512
w-u-jpr-- | atmega2560_min.hex | 248 | 256 | 248 | 256
weu-jPr-c | atmega2560_ur.hex | 386 | 512 | 386 | 512
wes-Vpr-c | atmega2560_v3.hex | 642 | 768 | 654 | 768
wes-hpr-c | atmega32.hex | 436 | 512 | 442 | 512
wes-hpr-c | atmega328p.hex | 446 | 512 | 446 | 512
weu-jPr-c | atmega328p_7875khz_swio_ur.hex | 380 | 384 | 380 | 384
weu-jPr-c | atmega328p_8000khz_swio_ur.hex | 378 | 384 | 378 | 384
weu-jPr-c | atmega328p_8125khz_swio_ur.hex | 376 | 384 | 376 | 384
weu-jPr-c | atmega328p_8mhz_ur.hex | 378 | 384 | 378 | 384
weudhprac | atmega328p_adur.hex | 510 | 512 | 510 | 512
w-u-jpra- | atmega328p_amin.hex | 254 | 256 | 254 | 256
weu-jPrac | atmega328p_autobaud_ur.hex | 376 | 384 | 376 | 384
weudhpr-c | atmega328p_dur.hex | 494 | 512 | 494 | 512
wes-hpr-c | atmega328p_led9_50Hz_fp9.hex | 468 | 512 | 468 | 512
w-u-jpr-- | atmega328p_min.hex | 252 | 256 | 252 | 256
weu-jPr-c | atmega328p_ur.hex | 360 | 384 | 360 | 384
weu-jPr-c | atmega328p_ur_testing.hex | 376 | 384 | 376 | 384
w-u-jpr-- | atmega32_min.hex | 244 | 256 | 246 | 256
weu-jPr-c | atmega32_ur.hex | 356 | 384 | 356 | 384
w-u-jpr-- | atmega644p_min.hex | 250 | 256 | 250 | 256
wes-hpr-c | atmega8.hex | 434 | 512 | 440 | 512
wes-hpr-c | atmega88.hex | 446 | 512 | 446 | 512
w-u-jpr-- | atmega88_min.hex | 252 | 256 | 252 | 256
weu-jpr-- | atmega88_ur.hex | 310 | 320 | 310 | 320
w-u-hpr-- | atmega8_min.hex | 244 | 256 | 246 | 256
weu-jpr-- | atmega8_ur.hex | 314 | 320 | 314 | 320
weudjpr-c | attiny167_dur.hex | 502 | 512 | 502 | 512
w-u-jpr-- | attiny167_min.hex | 252 | 256 | 252 | 256
weu-jPr-c | attiny167_ur.hex | 374 | 384 | 374 | 384
w-u-jPr-- | attiny2313_min.hex | 224 | 224 | 222 | 256
w-u-jPr-- | attiny84_min.hex | 256 | 256 | 276 | 320
w-u-jPr-- | attiny85_min.hex | 256 | 256 | 276 | 320
weu-jPr-c | attiny85_ur.hex | 378 | 384 | 378 | 384
wes-hpr-c | diecimila.hex | 446 | 512 | 446 | 512
weu-jPr-c | digispark.hex | 378 | 384 | 378 | 384
weu-jPr-c | digisparkpro.hex | 374 | 384 | 374 | 384
wes-hpr-c | jeenode.hex | 446 | 512 | 446 | 512
weu-jPr-c | lilypad_ur.hex | 378 | 384 | 378 | 384
weu-jpr-c | luminet_baud9600_ur.hex | 374 | 384 | 374 | 384
wes-hpr-c | moteino.hex | 446 | 512 | 446 | 512
weudhpr-c | moteino_cs8_dur.hex | 486 | 512 | 486 | 512
wes-hpr-c | pro_16mhz.hex | 446 | 512 | 446 | 512
weu-jPr-c | pro_16mhz_ur.hex | 360 | 384 | 360 | 384
wes-hpr-c | pro_20mhz.hex | 442 | 512 | 442 | 512
weu-jPr-c | pro_20mhz_ur.hex | 356 | 384 | 356 | 384
wes-hpr-c | pro_8mhz.hex | 464 | 512 | 464 | 512
weu-jPr-c | pro_8mhz_ur.hex | 378 | 384 | 378 | 384
wes-hpr-c | promini_led13.hex | 446 | 512 | 446 | 512
wes-hpr-c | promini_led9.hex | 446 | 512 | 446 | 512
wes-hpr-c | timeduino.hex | 446 | 512 | 446 | 512
weudhpr-c | timeduino_cs8_dur.hex | 486 | 512 | 486 | 512
wes-hpr-c | urclock.hex | 446 | 512 | 446 | 512
weudhpr-c | urclock_cs8_dur.hex | 486 | 512 | 486 | 512

mcuee, 2022-11-26:
@stefanrueger 

In summary, I think there is nothing to be changed on your side. In case users want to build under Windows or macOS, they can refer to my experiences here. I am happy to help as well if other users encountered problems under macOS or Windows.

stefanrueger, 2022-11-27:
@mcuee Thank you for all your hard work exploring ways to improve the urboot building process. When I get a minute I will put the results of your exploration into a "how-to-build" page; your work *will* be used to good effect.