stefanrueger/urboot #11: hexls issue with optiboot ATtinyCore

mcuee, 2022-12-10:
Today I encountered some issues from the hex files generated by ATtinyCore. I have no issues with other optiboot bootloaders so far, like those from optiboot_x (MegaCoreX and megaTinyCore) and DxCore.

https://github.com/SpenceKonde/ATTinyCore/tree/v2.0.0-devThis-is-the-head-submit-PRs-against-this/avr/bootloaders/optiboot/source

```
$ ./hexls optiboot_attiny88_8000000UL.hex
Argument "--" isn't numeric in addition (+) at ./hexls line 221.
Argument "----h--" isn't numeric in addition (+) at ./hexls line 221.
610 640 u7.3 0 optiboot_attiny88_8000000UL.hex

$ cat optiboot_attiny88_8000000UL.hex
:101D800001C025C1112484B7882379F0982F9A7057
:101D9000923059F081FF03C094B7977F94BF282EEB
:101DA00080E000D1E1E1FF27099485E08093810084
:101DB0008EE0F8D0259A569A86E028E13EEF91E031
:101DC000309385002093840096BBB09BFECF1D9A74
:101DD000A8958150A9F793E0D92ECC24C39425E08F
:101DE000922E32ECE32E3EE0F32E8F2D806C882E67
:101DF000C5D0813479F4C2D0182FDAD0123811F45A
:101E000080E004C08BE3113809F083E0A8D080E1C2
:101E1000A6D0EECF823419F484E1D2D0F8CF853445
:101E200011F485E0FACF853541F4A8D0C82FA6D0AB
:101E3000D82FCC0FDD1FBCD0EACF863519F484E053
:101E4000BFD0DECF843609F052C098D097D0782E1C
:101E500095D0682E00E011E05801EFEFAE1ABE0AEF
:101E60008DD0F801808385017A10F6CFA1D020971C
:101E7000F1F430910001309384012091010120930D
:101E80008501E0920001809201014091220140937E
:101E900086014091230140938701832F922F419720
:101EA00080932201892F8F70806C80932301F5E449
:101EB0006F1201C0FFCFFE01D7BEE89507B600FC48
:101EC000FDCFFE01A0E0B1E0CD0102962D913C9145
:101ED0000901C7BEE89511243296DC017812F4CFCF
:101EE000FE0197BEE89507B600FCFDCF90CF843782
:101EF00039F544D043D0B82E41D05AD08E010115C7
:101F0000110549F4809184012AD0BA940F5F1F4FC4
:101F1000B110F5CF7CCF0130110519F48091850106
:101F2000F3CF0232110519F480918601EDCF03320F
:101F3000110519F480918701E7CFF8018491E4CF6E
:101F4000853739F435D08EE10AD083E908D081E1B4
:101F50005DCF813509F06FCF88E024D06CCF2AE0C7
:101F600030E08095089410F45E9802C05E9A0000FC
:101F700015D014D086952A95B1F70895A89529E033
:101F800030E04F99FECF0AD009D008D088944F99FD
:101F900008942A9511F08795F7CF08959AE29A95BB
:101FA000F1F70895E0E6F0E098E1908380830895EA
:101FB000E5DF803219F088E0F5DFFFCF84E1CFCF95
:101FC000CF93C82FDBDFC150E9F7CF91F1CFFC01F0
:101FD0000A0167BFE895112407B600FCFDCF0895FC
:021FFE00003BA6
:0400000300001D805C
:00000001FF

```

stefanrueger, 2022-12-11:
> issues from the hex files generated by ATtinyCore

hexls is predominantly meant to work for urclock bootloaders (and soe "classic" optiboot bootloaders). It's likely hexls misreads the top byte in this particular bootloader and thinks it's an urboot 7.3 bootloader. Notwithstanding this also found a string concatenation error. hexls mistakenly used `+=` instead of `.=`

Just pushed a commit for the latter.

mcuee, 2022-12-11:
It is okay now. I think it serves my purposes already even though it may mistakenly show the hex as urboot 7.3, at least there are no error messages already.

```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_attinycore/hex$ ./hexls optiboot_attiny88_8000000UL.hex 
594 640 u7.3 ----h---- optiboot_attiny88_8000000UL.hex
```

mcuee, 2022-12-11:
I will close this issue now.

closed 2022-12-11 by mcuee