stefanrueger/urboot #12: USB urboot for ATmega32U4 and similar

mcuee, 2022-12-19:
Caterina bootloader (USB AVR109) is popular for ATmega32U4 based boards (eg: Arduino Leonardo and Arduino Micro). However they are a bit troublesome to work with avrdude (1200bps touch).

Please refer to the following discussion in avrdude.
* https://github.com/avrdudes/avrdude/issues/788

It may be a good idea to port urboot to ATmega32U4 (and similar) with USB to Serial support.

stefanrueger, 2022-12-19:
> USB to Serial support

I have insufficient experience with USB to serial. Urboot assumes there is a serial line to talk to, and the rest is abstracted away in the uploader (eg, AVRDUDE). Are you talking about parts that have native USB support in the AVR and that would use something else than an USART for comms? That kind of stuff would need a PR from someone else who has the experience.

Other than that the [Avrdude issue #788](https://github.com/avrdudes/avrdude/issues/788) is independent from bootloading itself, so invalid here.

mcuee, 2022-12-20:
> Are you talking about parts that have native USB support in the AVR and that would use something else than an USART for comms

Basically the idea is the same as Caterina bootloader (USB to Serial + AVR109) so that we do not need a seperate USB to Serial chip to use urboot (USB to Serial + urboot) since the USB AVR can act as a USB to Serial chip ifself by implementing USB CDC-ACM FW.

Reference:
https://github.com/arduino/ArduinoCore-avr/tree/master/bootloaders/caterina

mcuee, 2022-12-20:
There is a potential issue of USB VID/PID to use though.

stefanrueger, 2023-03-01:
I have carefully looked at the USB interface of these parts in the data sheet. It's some 30 pages of dense description and I imagine it needs *a lot* of code to directly read/write USB serial data. I am tempted to predict that this will never happen. Urboot is designed to be compact (it's roughly half the size of optiboot), so that's not the type of project to start spending a lot of code in the bootloader for USB to serial conversion. That really is better done by an USB cable/chip.

mcuee, 2023-03-01:
Fair enough. I agree this is out of scope.

closed 2023-03-01 by mcuee