stefanrueger/urboot #14: Provide options to build urboot for other serial port

mcuee, 2022-12-22:
It would be great to provide urboot hex files for non-default serial port. I think this is one of the major missing feature from urboot.

As per the previous discussions in avrdude, it may not be so easy to do this automatically though.

Maybe @MCUdude has some ideas since he has done this in [optiboot_flash](https://github.com/MCUdude/optiboot_flash).

mcuee, 2022-12-22:
Features like automatically detect which serial port to use may be even more troublesome and may not be worth the efforts.
* https://github.com/Optiboot/optiboot/issues/285

stefanrueger, 2022-12-22:
It's certainly possible to compile bootloaders for USART1, USART2, ... for fixed baud rates: When you compile an urboot bootloader you can set which USART you want the comms to be on: Just specify `UARTNUM=x` when using `make` or `-DUARTNUM=x` when compiling... (x=0, 1, 2, ...). 

However, for autobaud `urboot.c` needs to know the RX line of the chosen USART to measure the baud rate before initialising the USART. The table here [here](https://github.com/avrdudes/avrdude/pull/1171#issuecomment-1315228253) needs filling in. The atdf files are not great for this. Maybe someone can help?

(As for the [Optiboot Issue #285](https://github.com/Optiboot/optiboot/issues/285) I don't know of a valid use case, and cannot imagine one, for the bootloader "automatically" scanning where communications come from. In my world, bootloaders should utilise as few I/O lines as possible and these should be known and fixed; otherwise if the bootloader starts twiggling all sorts of I/O lines at each reset, the projects implemented on the board might do strange things while the I/O lines are scanned: just imagine one of the I/O lines opens the gate of the attached alligator compound... this does not bear thinking about :)

mcuee, 2022-12-22:
The table is copied here fore reference.

> I actually don't even have a table for now many USARTS any one part has (beyond 0 or > 0).
> 
> Here is an example of what would be good:

| Part | N<sub>UART</sub> || RXD0 | TXD0 ||  RXD1 | TXD1 || RXD2 | TXD2 || RXD3 | TXD3 |
| --:  |:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|
|attiny167| 0  || | || | || | || | |
|atmega328p | 1 || PD0 | PD1|| | || | || | |
|atmega328pb| 2 || PD0 | PD1|| PB4 | PB3 || | || | |
|atmega2560| 4 || PE0 | PE1 || PD2 | PD3 || PH0 | PH1 || PJ0 | PJ1 |

> Here is what I have in machine-readable format (you can tell my xml-fu with atdfs is waning). And there are likely some mistakes in that table, too :(

| Part | N<sub>UART</sub> || RXD0 | TXD0 ||  RXD1 | TXD1 || RXD2 | TXD2 || RXD3 | TXD3 |
| --:  |:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|:-:|
|ATtiny4|0|| | || | || | || | |
|ATtiny5|0|| | || | || | || | |
|ATtiny9|0|| | || | || | || | |
|ATtiny10|0|| | || | || | || | |
|ATtiny20|0|| | || | || | || | |
|ATtiny40|0|| | || | || | || | |
|ATtiny102| >0 ||PB3|PB2|| | || | || | |
|ATtiny104| >0 ||PB3|PB2|| | || | || | |
|ATtiny11|0|| | || | || | || | |
|ATtiny12|0|| | || | || | || | |
|ATtiny13|0|| | || | || | || | |
|ATtiny13A|0|| | || | || | || | |
|ATtiny15|0|| | || | || | || | |
|ATtiny22| ?? || | || | || | || | |
|ATtiny24|0|| | || | || | || | |
|ATtiny24A|0|| | || | || | || | |
|ATtiny25|0|| | || | || | || | |
|ATtiny26|0|| | || | || | || | |
|ATtiny28|0|| | || | || | || | |
|ATtiny43U|0|| | || | || | || | |
|ATtiny44|0|| | || | || | || | |
|ATtiny44A|0|| | || | || | || | |
|ATtiny45|0|| | || | || | || | |
|ATtiny48|0|| | || | || | || | |
|ATtiny84|0|| | || | || | || | |
|ATtiny84A|0|| | || | || | || | |
|ATtiny85|0|| | || | || | || | |
|ATtiny87|0|| | || | || | || | |
|ATtiny88|0|| | || | || | || | |
|ATtiny167|0|| | || | || | || | |
|ATtiny261|0|| | || | || | || | |
|ATtiny261A|0|| | || | || | || | |
|ATtiny441| >0 ||PA4|PA5|| | || | || | |
|ATtiny461|0|| | || | || | || | |
|ATtiny461A|0|| | || | || | || | |
|ATtiny828| >0 ||PC2|PC3|| | || | || | |
|ATtiny828R| ?? || | || | || | || | |
|ATtiny841| >0 ||PA4|PA5|| | || | || | |
|ATtiny861|0|| | || | || | || | |
|ATtiny861A|0|| | || | || | || | |
|ATtiny1634| >0 ||PA7|PB0|| | || | || | |
|ATtiny1634R| ?? || | || | || | || | |
|ATtiny2313| >0 ||PD0|PD1|| | || | || | |
|ATtiny2313A| >0 ||PD0|PD1|| | || | || | |
|ATtiny4313| >0 ||PD0|PD1|| | || | || | |
|ATmega8| >0 ||PD0|PD1|| | || | || | |
|ATmega8A| >0 ||PD0|PD1|| | || | || | |
|ATmega8HVA|0|| | || | || | || | |
|ATmega8U2| >0 ||PD2|PD3|| | || | || | |
|ATmega16| >0 ||PD0|PD1|| | || | || | |
|ATmega16A| >0 ||PD0|PD1|| | || | || | |
|ATmega16HVA|0|| | || | || | || | |
|ATmega16HVB|0|| | || | || | || | |
[... 294 lines cut ...]
|AVR32DB48| >0 ||PA1|PA0|| | || | || | |
|AVR32EA48| ?? || | || | || | || | |
|AVR64DD14| >0 ||PA1|PA0|| | || | || | |
|AVR64DD20| >0 ||PA1|PA0|| | || | || | |
|AVR64DA28| >0 ||PA1|PA0|| | || | || | |
|AVR64DB28| >0 ||PA1|PA0|| | || | || | |
|AVR64DD28| >0 ||PA1|PA0|| | || | || | |
|AVR64EA28| >0 ||PA1|PA0|| | || | || | |
|AVR64DA32| >0 ||PA1|PA0|| | || | || | |
|AVR64DB32| >0 ||PA1|PA0|| | || | || | |
|AVR64DD32| >0 ||PA1|PA0|| | || | || | |
|AVR64EA32| >0 ||PA1|PA0|| | || | || | |
|AVR64DA48| >0 ||PA1|PA0|| | || | || | |
|AVR64DB48| >0 ||PA1|PA0|| | || | || | |
|AVR64EA48| >0 ||PA1|PA0|| | || | || | |
|AVR64DA64| >0 ||PA1|PA0|| | || | || | |
|AVR64DB64| >0 ||PA1|PA0|| | || | || | |
|AVR128DA28| >0 ||PA1|PA0|| | || | || | |
|AVR128DB28| >0 ||PA1|PA0|| | || | || | |
|AVR128DA32| >0 ||PA1|PA0|| | || | || | |
|AVR128DB32| >0 ||PA1|PA0|| | || | || | |
|AVR128DA48| >0 ||PA1|PA0|| | || | || | |
|AVR128DB48| >0 ||PA1|PA0|| | || | || | |
|AVR128DA64| >0 ||PA1|PA0|| | || | || | |
|AVR128DB64| >0 ||PA1|PA0|| | || | || | |

mcuee, 2023-03-01:
Nice one. I will carry out some tests and report here soon, probably over this weekend.

stefanrueger, 2023-03-01:
The problem is not where the USART registers are, but where the Port for the RX line is. It is this loop https://github.com/stefanrueger/urboot/blob/74ecad651111878a53752a03896074d7230ce12f/src/urboot.c#L1956-L1957 that needs bit access (using the `sbic` opcode) to the `PIN` of the Port of the RX line associated to USART2. And the PIN of Port H is *way* outside I/O space [0x20, 0x3f]. One loop iteration takes 5 cycles (mapping neatly to the 5 low bit that are measured after the start bit) and cannot afford to be substantially longer...

stefanrueger, 2023-03-01:
Maybe I should have clarified: the *bit-addressable I/O space* is an address region of only 32 bytes (normally from 0x20 onwards, but in some parts from 0 onwards). Only in these 32 bytes can every single bit be individually addressed by the `sbic`, `sbic`, `sbi`, `cbi` opcodes.

stefanrueger, 2023-03-02:
So, I have now refined the UART capabilities, and even cater for ATmega161's single byte that the two different msb-4 bit share between the two 12-bit BRR registers of UART0 and UART1. If anyone has an ATmega161, that's a good edge case to test (slow baud rate with high F_CPU on UART1).

mcuee, 2023-03-05:
@stefanrueger 

Tested the ATmega2560 115200bps bootloader for UART0/1/2/3 and they all work well.
https://github.com/stefanrueger/urboot.hex/tree/main/boards/mega-r3/fcpu_16mhz/115200_bps

Test results here.
* https://github.com/avrdudes/avrdude/issues/1262#issuecomment-1455022993

I use the avrdude patch from Marius for the testing of UART1/2/3 since I do not have the capacitors with me.

stefanrueger, 2023-03-15:
Thanks for testing @mcuee!

So, I trust this issue can now be closed?

mcuee, 2023-03-15:
Yes this can be closed.

closed 2023-03-15 by mcuee