stefanrueger/urboot #15: github action CI

mcuee, 2022-12-22:
It may be a good idea to create github action CI for urboot, probably just for Linux as the toolchain in the repo is only for Linux.

One possibility is to use Docker container in the CI.

stefanrueger, 2023-03-03:
The urboot project only works under Linux and can easily be "checked" by executing `make all`, which generates a variety of bootloaders. So, CI is less of an issue for this project. I would like to provide a working docker container for the ability to `make` bootloaders in other environments (eg, Mac, Windows) but lack the drive, skills and test beds (Macs, Windows) to do so.

mcuee, 2023-03-03:
Docker container will be good. I do not know much but if you come out with the Docker Compose file, I should be able to carry out the test under Mac and Windows.

PR #31 (2024-03-30): add dockerfile for container builds
Adds a small Dockerfile definition to create a Linux build environment on non-linux machines + provides documentation in the README to build w/ zero setup.

Tested on aarch64 macOS (using Docker to simulate x86-64 linux for precompiled AVR toolchain) and x86-64 linux.

From the repo root directory:
```
docker run --platform linux/amd64 -v "$(pwd)/src":/src --rm -it $(docker build -q .) all
```
is equivalent to `make all`.

(closes #15)
closed 2024-03-30 by stefanrueger