stefanrueger/urboot #16: More documentations

mcuee, 2022-12-22:
I think the current documentation is already pretty good. But it 

1) What is urboot -- README.md

2) urprotocol -- urprotocol.md

3) How to choose urboot bootloader options -- -- howtoselect.md

4) Compiling urboot -- more details to be added, some simple explanations about the usage of urboot-gcc and hexls would be nice.

5) How urboot works -- I am not so sure if something like the following is good to have or not.
https://github.com/Optiboot/optiboot/wiki/HowOptibootWorks

6) FAQ -- to be added

stefanrueger, 2022-12-31:
@mcuee Gave it a shot. I think that now the documentation is as good as it gets.

closed 2022-12-31 by stefanrueger

mcuee, 2022-12-31:
> @mcuee Gave it a shot. I think that now the documentation is as good as it gets.

Thanks. Now it seems to be pretty comprehensive from a quick glance.

The only missing thing is to mention `MH Tiny` in https://github.com/stefanrueger/urboot/blob/main/README.md

stefanrueger, 2022-12-31:
| The only missing thing is to mention

Well there are tons of other boards in the [boards](https://github.com/stefanrueger/urboot.hex/tree/main/boards) directory that are not mentioned. We have to draw a line with the examples somewhere :smile:

mcuee, 2022-12-31:
No issues. I understand.