stefanrueger/urboot #17: Seperate out the big collection of bootloader hex file into a seperate repo

mcuee, 2022-12-23:
This may be a personal taste but I feel that it will be better to seperate the  big collection of hex files in the following directory into a seperate repo, say urboot_bin.
https://github.com/stefanrueger/urboot/tree/main/bootloaders

mcuee, 2022-12-23:
Related discussions:
* https://github.com/Optiboot/optiboot/issues/355#issuecomment-1364387824
* https://github.com/MCUdude/optiboot_flash/issues/9

stefanrueger, 2022-12-24:
Yes, that's a matter of taste; I don't think this is a priority

mcuee, 2022-12-24:
In this case, I will close the issue.

closed 2022-12-24 by mcuee

stefanrueger, 2022-12-31:
I had a change of heart and moved the bootloaders into a different repo [urboot.hex](https://github.com/stefanrueger/urboot.hex)

mcuee, 2022-12-31:
> I had a change of heart and moved the bootloaders into a different repo [urboot.hex](https://github.com/stefanrueger/urboot.hex)

Wonderful!

stefanrueger, 2022-12-31:
The reason was that `git status` was getting a bit slow. Now, I am likely to squash merge this repo, so that less than 0.6 GB get cloned :open_mouth: (erasing all the previously published bootloaders from the git history)