stefanrueger/urboot #19: Urboot for ATTiny13/A

MCUdude, 2023-03-01:
Unlike other AVRs, the internal oscillator in the ATtiny13/A runs at 9.6 MHz and is dividable down to 4.8 MHz, 1.2MHz, and 600kHz.
It would be great if these existed pre-compiled bootloaders for these clock speeds!

MCUdude, 2023-03-01:
That would be excellent! Is 250000 baud the fastest "standard" baudrate the ATtiny13/A can handle without too much error?

stefanrueger, 2023-03-01:
> Is 250000 baud the fastest "standard" baudrate the ATtiny13/A can handle without too much error?

Yes.

The ATtiny13 does not have a hardware UART. So, F_CPU/baud rate is the budget of clock cycles that the software has for delaying half a bit, sensing the RX bit, then delaying half a bit. 9.6 MHz/250 kbaud gives us 38.4 cycles. That should just about work, but I doubt one can do SWIO with 460800 baud (only 20 cycles for a complex task).

BTW, 38.4 cycles also means an intrinsic quantisation error of 0.4/38.4 ~ 1%, which is OK assuming the internal oscillator does not add more than about 1.5% of frequency deviation. Smaller baud rates have a better quantisation error: for example, at 115200 baud there is a budget of 83.333 cycles where the quantisation error is 0.33/83.33 = 0.4 %, so you can afford the internal oscillator to be off by max 2%.

stefanrueger, 2023-03-02:
OK, done: see, eg, [ATtiny13A](https://github.com/stefanrueger/urboot.hex/tree/main/mcus/attiny13a) bootloaders

MCUdude, 2023-03-03:
Thanks! I'll be away for the weekend, but I'll give it a try next week when I'm back home

stefanrueger, 2023-03-03:
I checked the code for the max baud rate with (my!) SWIO code. It is F_CPU/29 for classic parts with 16-bit PC and F_CPU/33 for classic parts with 22-bit PC (ATmega2560, eg). XMEGA parts can achieve a slightly higher baud rates: F_CPU/28 and F_CPU/32, respectively. This is owing to subtle changing in the clock cycles for opcodes.

BTW, is there a good documentation for the timings of newer parts? Some of these have `__AVR_XMEGA__` defined in the gcc-avr pre-processor: `atmega1608` `atmega1609` `atmega3208` `atmega3209` `atmega4808` `atmega4809` `atmega808` `atmega809` `attiny1604` `attiny1606` `attiny1607` `attiny1614` `attiny1616` `attiny1617` `attiny202` `attiny204` `attiny212` `attiny214` `attiny3214` `attiny3216` `attiny3217` `attiny402` `attiny404` `attiny406` `attiny412` `attiny414` `attiny416` `attiny417` `attiny804` `attiny806` `attiny807` `attiny814` `attiny816` `attiny817` 

Do these parts have XMEGA timings? @SpenceKonde @MCUdude @mcuee

mcuee, 2023-03-05:
I am not an AVR expert but I tend to think they have the xmega timing from what I read.
https://en.wikipedia.org/wiki/ATtiny_microcontroller_comparison_chart
https://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html
https://github.com/SpenceKonde/megaTinyCore#megatinycore---background-and-basics

mcuee, 2023-03-05:
The Urboot for ATtiny13A seems to work fine.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue_1262_mod -c usbasp -p t13a
 -U flash:v:.\urboot_attiny13a_9mhz6_115200bps_swio_rxb0_txb1_ee_lednop_fr_ce_ur_vbl.hex:i -qq && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue_1262_mod -c urclock -P COM7
 -p t13a -xshowall -q

avrdude_issue_1262_mod: AVR device initialized and ready to accept instructions
000075716854 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 
weu-jpr-c vector 4 (EE_RDY) ATtiny13A

```

mcuee, 2023-03-05:
Same for normal mode. EEPROM writing is okay but not flash writing.

```
PS C:\work\avr\avrdude_test\avrdude_bin> cat .\entest_64B.eep
:020000040000FA
:2000000054686520717569636B2062726F776E20666F78206A756D7073206F76657220740E
:200020006865206C617A7920646F670A54686520717569636B2062726F776E20666F78207C
:00000001FF

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue_1262_mod -c urclock -P COM7
 -p t13a -U eeprom:w:entest_64B.eep:i

avrdude_issue_1262_mod: AVR device initialized and ready to accept instructions
avrdude_issue_1262_mod: device signature = 0x1e9007 (probably t13a)
avrdude_issue_1262_mod: reading input file entest_64B.eep for eeprom
                        with 64 bytes in 1 section within [0, 0x3f]
                        using 16 pages and 0 pad bytes
avrdude_issue_1262_mod: writing 64 bytes eeprom ...

Writing | ################################################## | 100% 0.24 s

avrdude_issue_1262_mod: 64 bytes of eeprom written
avrdude_issue_1262_mod: verifying eeprom memory against entest_64B.eep

Reading | ################################################## | 100% 0.06 s

avrdude_issue_1262_mod: 64 bytes of eeprom verified

avrdude_issue_1262_mod done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> cat .\zeros_t13a_test.hex
:020000040000FA
:200100000000000000000000000000000000000000000000000000000000000000000000DF
:200120000000000000000000000000000000000000000000000000000000000000000000BF
:2001400000000000000000000000000000000000000000000000000000000000000000009F
:2001600000000000000000000000000000000000000000000000000000000000000000007F
:2001800000000000000000000000000000000000000000000000000000000000000000005F
:2001A00000000000000000000000000000000000000000000000000000000000000000003F
:2001C00000000000000000000000000000000000000000000000000000000000000000001F
:2001E0000000000000000000000000000000000000000000000000000000000000000000FF
:00000001FF

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue_1262_mod -c urclock
 -P COM7 -p t13a -U .\zeros_t13a_test.hex

avrdude_issue_1262_mod: AVR device initialized and ready to accept instructions
avrdude_issue_1262_mod: device signature = 0x1e9007 (probably t13a)
avrdude_issue_1262_mod: Note: flash memory has been specified, an erase cycle will be performed.
                        To disable this feature, specify the -D option.
avrdude_issue_1262_mod: erasing chip
avrdude_issue_1262_mod: reading input file .\zeros_t13a_test.hex for flash
                        with 256 bytes in 1 section within [0x100, 0x1ff]
                        using 8 pages and 0 pad bytes
avrdude_issue_1262_mod: preparing flash input for device bootloader
avrdude_issue_1262_mod: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.04 s

avrdude_issue_1262_mod: 256 bytes of flash written
avrdude_issue_1262_mod: verifying flash memory against .\zeros_t13a_test.hex

Reading |                                                    | 0% 
0.00 s avrdude_issue_1262_mod: en passant forcing reset vector to point to vector bootloader
Reading | ################################################## | 100% 0.05 s

avrdude_issue_1262_mod warning: verification mismatch
                       device 0xff != input 0x00 at addr 0x0100 (error)
avrdude_issue_1262_mod error: verification mismatch

avrdude_issue_1262_mod done.  Thank you.
```

stefanrueger, 2023-03-05:
@mcuee Thanks for testing! Looks like you need to enable `SELFPRGEN` in `hfuse`
```
avrdude> dump hfuse
0000  fb                                                |.               |
 ```

stefanrueger, 2023-03-05:
@mcuee And thanks for the pointers to exact timings.

mcuee, 2023-03-06:
> @mcuee Thanks for testing! Looks like you need to enable `SELFPRGEN` in `hfuse`
> 
> ```
> avrdude> dump hfuse
> 0000  fb                                                |.               |
> ```

That is it. Thanks for pointing out this apparent mistake. Now it works well.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c usbasp -p t13a -qqt
avrdude> dump lfuse
0000  3a                                                |:               |

avrdude> dump hfuse
0000  eb                                                |.               |

avrdude> quit

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 
weu-jpr-c vector 4 (EE_RDY) ATtiny13A

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -p t13a
 -U .\zeros_t13a_test.hex

avrdude_pr1309: AVR device initialized and ready to accept instructions
avrdude_pr1309: device signature = 0x1e9007 (probably t13a)
avrdude_pr1309: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_pr1309: erasing chip
avrdude_pr1309: reading input file .\zeros_t13a_test.hex for flash
                with 256 bytes in 1 section within [0x100, 0x1ff]
                using 8 pages and 0 pad bytes
avrdude_pr1309: preparing flash input for device bootloader
avrdude_pr1309: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.04 s

avrdude_pr1309: 256 bytes of flash written
avrdude_pr1309: verifying flash memory against .\zeros_t13a_test.hex

Reading | ################################################## | 100% 0.04 s

avrdude_pr1309: 256 bytes of flash verified

avrdude_pr1309 done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 2023-03-06 13.22 zeros_t13a_test.hex 512 store 97 meta 31 boot 384 u7.7 
weu-jpr-c vector 4 (EE_RDY) ATtiny13A
```

mcuee, 2023-03-06:
250,000 bps is also fine.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c usbasp -p t13a -qq
 -U .\urboot_attiny13a_9mhz6_250000bps_swio_rxb0_txb1_ee_lednop_fr_ce_ur_vbl.hex -qq && echo OK
OK

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -b 250000
 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 
weu-jPr-c vector 4 (EE_RDY) ATtiny13A

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -b 250000
 -p t13a -U .\zeros_t13a_test.hex

avrdude_pr1309: AVR device initialized and ready to accept instructions
avrdude_pr1309: device signature = 0x1e9007 (probably t13a)
avrdude_pr1309: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_pr1309: erasing chip
avrdude_pr1309: reading input file .\zeros_t13a_test.hex for flash
                with 256 bytes in 1 section within [0x100, 0x1ff]
                using 8 pages and 0 pad bytes
avrdude_pr1309: preparing flash input for device bootloader
avrdude_pr1309: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.04 s

avrdude_pr1309: 256 bytes of flash written
avrdude_pr1309: verifying flash memory against .\zeros_t13a_test.hex

Reading | ################################################## | 100% 0.02 s

avrdude_pr1309: 256 bytes of flash verified

avrdude_pr1309 done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -b 250000
 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 2023-03-06 13.22 zeros_t13a_test.hex 512 store 97 meta 31 boot 384 u7.7 
weu-jPr-c vector 4 (EE_RDY) ATtiny13A
```

stefanrueger, 2023-03-07:
> 250,000 bps is also fine.

That's good. Thanks for testing @mcuee. Amazing that the bootloader works on such a small flash/sram. The smallest b/loader is 256 bytes so can be used for rapid prototyping...

mcuee, 2023-03-07:
@MCUdude 

Haha, now you may have a chance to add urboot to MicroCore.
https://github.com/MCUdude/MicroCore

MCUdude, 2023-03-07:
Adding Urboot to MicroCore is definitly something I'll consider!

The only problem is that when I designed the "MicroCore hardware", I decided to keep compatibility with the existing ATtinyCore and its bootloader implementation, so that ATtiny85's could be programmed usimmg my board without modifications.

The quirk here is that ATtinyCore and MicroCore use pin PB0 for TX, and PB1 for RX. The opposite of what the pre-compiled urboot binaries does.

mcuee, 2023-03-07:
> The quirk here is that ATtinyCore and MicroCore use pin PB0 for TX, and PB1 for RX. The opposite of what the pre-compiled urboot binaries does.

That should be a easy change.

MCUdude, 2023-03-07:
It would be awesome to have a menu in the Arduino tools menu like this. For rapid prototyping, a bootloader is very convenient

```
Bootloader:
  * No bootloader
  * RX = PB1, TX = PB0
  * RX = PB0, TX = PB1
```

MCUdude, 2023-03-07:
I'll look more into this tonight.
The board has an onboard LED connected to pin PB2 (active high), but it would only make sense to add support for this if the compiled binary occupies 256 bytes or less.

stefanrueger, 2023-03-07:
> the compiled binary occupies 256 bytes or less

You can judge this by checking out if there is a `_lednop` bootloader that is within 256 bytes. This has "placeholder nops" for where LED twiddling goes once you know the pin. If there are only non-`_lednop` bootloaders within 256 bytes, then the bootloaders can be shortened by 4 bytes if needed  through the tricks above.

|  I can only assume that a hardware reset also resets the internal WD timer to 0...

That is actually highly likely come to think of that. Most AVR will have a fuse setting that switches on the WD timer by default. In order for these to work the WD timer needs to be reset to 0 on external (hard) reset. Otherwise there would be an unknown period for the first WDTO after external reset, which isn't acceptable. I believe the function that configures the WDTO in urboot is the *only* aspect where optiboot has shorter code (as optoboot drops the wdr)

MCUdude, 2023-03-07:
It looks like the lednop bootloaders for ATtiny13/A are bigger than 256 bytes, so I think it's best with a bootloader that doesn't flash the LED while programming, but can instead fit within 256 bytes.
https://github.com/stefanrueger/urboot.hex/tree/main/mcus/attiny13a/fcpu_9mhz6/115200_bps

(Ralph Doncaster actually created a 64-byte (!) sized bootloader, [picoboot](https://github.com/nerdralph/picoboot). I'm not sure how "safe" this is compared to Urboot, but it's still impressive in its own right!)

@stefanrueger could you add MicroCore as another "core" in the urboot repo that contains pre-compiled bootloaders for ATtiny13A with RXD=PB0/TXD=PB1 and RXD=PB1/TXD=PB0? I see no reason why _not_ to add a bootloader option to MicroCore when a proper, failsafe bootloader can fit within 256 bytes!

MCUdude, 2023-03-07:
I've just pushed a commit to the [MicroCore repo](https://github.com/MCUdude/MicroCore) where I've added Urboot as an option in the Arduino tools menu. Using Urboot with the 9.6 MHz internal oscillator works like a charm, but I'm having issues getting the 4.8MHz, 1.2MHz, and 600kHz options to work. @mcuee can you test it on your hardware? If you want to use Arduino IDE directly, you'll have to install MicroCore using the boards manager install, and then replace the installed core files with the one found in the repo on Github.

4.8 MHz option:
```
/Users/hans/Library/Arduino15/packages/MicroCore/tools/avrdude/7.1-arduino.1/bin/avrdude -C/Users/hans/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf -v -pattiny13a -curclock -b57600 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_151041/Blink.ino.hex:i 

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : urclock
         Overriding Baud Rate          : 57600
avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
avrdude urclock_getsync() warning: attempt 2 of 10: not in sync
avrdude urclock_getsync() warning: attempt 3 of 10: not in sync
avrdude urclock_getsync() warning: attempt 4 of 10: not in sync
avrdude urclock_getsync() warning: attempt 5 of 10: not in sync
avrdude urclock_getsync() warning: attempt 6 of 10: not in sync
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port /dev/cu.usbserial-1410

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected
```

1.2 MHz option:
```
/Users/hans/Library/Arduino15/packages/MicroCore/tools/avrdude/7.1-arduino.1/bin/avrdude -C/Users/hans/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf -v -pattiny13a -curclock -b19200 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_151041/Blink.ino.hex:i 

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : urclock
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13A
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
[... 42 lines cut ...]
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Library/Arduino15/packages/MicroCore/hardware/avr/2.2.1/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : urclock
         Overriding Baud Rate          : 9600
avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
avrdude urclock_getsync() warning: attempt 2 of 10: not in sync
avrdude urclock_getsync() warning: attempt 3 of 10: not in sync
avrdude urclock_getsync() warning: attempt 4 of 10: not in sync
avrdude urclock_getsync() warning: attempt 5 of 10: not in sync
avrdude urclock_getsync() warning: attempt 6 of 10: not in sync
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
An error occurred while uploading the sketch
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port /dev/cu.usbserial-1410

avrdude done.  Thank you.
```

stefanrueger, 2023-03-07:
If 9m6 Hz @ 115200 baud works like a charm, the so should 4m8 Hz @ 57600 baud. It's even the same bootloader bit for bit, just that the entire "world" is slower. Will be the same for 1m2 Hz @ 14400 baud with *exactly* the same bootloader, no need to recompile or reflash. Could you try these fuse settings to effect the correct F_CPU and then set the correct `-B <baud>` for avrdude. Sometimes I find that certain `-B ...` settings are not done (haven't gone to the bottom of that) but there must be a way to verify avrdude is setting the baud rate correctly.

> Ralph Doncaster actually created a 64-byte (!) sized bootloader

Yes, cool stuff. I read at the source and think I understand how it works. True, safety is all shifted to the uploader (but there is an eor check sum every 4 command bytes); there is no download; baud rate is a fixed function of F_CPU (no code left for delay loops to slow baud rate down), uploader needs to be able to create that baud rate, but yes, actually *very* cool stuff.

stefanrueger, 2023-03-08:
I am unhappy with the alphabetic order of the oscillator frequency directories in the urboot.hex github so thought of
```
autobaud
b_F_cpu__24m_Hz
c_F_cpu__22m1184_Hz
d_F_cpu__20m_Hz
e_F_cpu__18m432_Hz
f_F_cpu__16m_Hz
g_F_cpu__14m7456_Hz
h_F_cpu__12m_Hz
i_F_cpu__11m0592_Hz
j_F_cpu___9m6_Hz
k_F_cpu___8m_Hz
l_F_cpu___7m3728_Hz
m_F_cpu___4m8_Hz
n_F_cpu___4m_Hz
o_F_cpu___3m6864_Hz
p_F_cpu___2m_Hz
q_F_cpu___1m8432_Hz
r_F_cpu___1m2_Hz
s_F_cpu___1m_Hz
t_F_cpu_600k_Hz
```

Similarly for the baud rates
```
a_1000000_bps
b__500000_bps
c__460800_bps
d__250000_bps
e__230400_bps
f__125000_bps
g__115200_bps
h___76800_bps
i___57600_bps
j___38400_bps
k___28800_bps
l___19200_bps
m___14400_bps
n____9600_bps
o____7200_bps
p____4800_bps
q____3600_bps
r____2400_bps
s____1200_bps
```

Do I miss some?

stefanrueger, 2023-03-08:
Could also do

```
autobaud
b_F_cpu_24.0000_MHz
c_F_cpu_22.1184_MHz
d_F_cpu_20.0000_MHz
e_F_cpu_18.4320_MHz
f_F_cpu_16.0000_MHz
g_F_cpu_14.7456_MHz
h_F_cpu_12.0000_MHz
i_F_cpu_11.0592_MHz
j_F_cpu__9.6000_MHz
k_F_cpu__8.0000_MHz
l_F_cpu__7.3728_MHz
m_F_cpu__4.8000_MHz
n_F_cpu__4.0000_MHz
o_F_cpu__3.6864_MHz
p_F_cpu__2.0000_MHz
q_F_cpu__1.8432_MHz
r_F_cpu__1.2000_MHz
s_F_cpu__1.0000_MHz
t_F_cpu__0.6000_MHz
```
Probably even neater (thinking out loud here)

mcuee, 2023-03-08:
> If 9m6 Hz @ 115200 baud works like a charm, then so should 4m8 Hz @ 57600 baud. It's even the same bootloader bit for bit, just that the entire "world" is slower.

@MCUdude and @stefanrueger 

Yes that is the case for me.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 
weu-jpr-c vector 4 (EE_RDY) ATtiny13A

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a -qqt
avrdude> dump lfuse
0000  3a                                                |:               |

avrdude> write lfuse 0 0x39
avrdude> quit
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7
 -b 57600 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 0000-00-00 00.00  application 0 store 0 meta 0 boot 384 u7.7 
weu-jpr-c vector 4 (EE_RDY) ATtiny13A

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -b 57600
 -p t13a -U .\zeros_t13a_test.hex

avrdude_pr1309: AVR device initialized and ready to accept instructions
avrdude_pr1309: device signature = 0x1e9007 (probably t13a)
avrdude_pr1309: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_pr1309: erasing chip
avrdude_pr1309: reading input file .\zeros_t13a_test.hex for flash
                with 256 bytes in 1 section within [0x100, 0x1ff]
                using 8 pages and 0 pad bytes
avrdude_pr1309: preparing flash input for device bootloader
avrdude_pr1309: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.04 s

avrdude_pr1309: 256 bytes of flash written
avrdude_pr1309: verifying flash memory against .\zeros_t13a_test.hex

Reading | ################################################## | 100% 0.03 s

avrdude_pr1309: 256 bytes of flash verified

avrdude_pr1309 done.  Thank you.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1309 -c urclock -P COM7 -b 57600 -p t13a -xshowall

avrdude_pr1309: AVR device initialized and ready to accept instructions
000020656854 2023-03-06 13.22 zeros_t13a_test.hex 512 store 97 meta 31 boot 384 u7.7
 weu-jpr-c vector 4 (EE_RDY) ATtiny13A

```

mcuee, 2023-03-08:
@MCUdude 

I use your git master zip file to replace the MicroCore 2.2.1 release and change the avrdude version from 7.1-arduino.1 to avrdude PR 1309 (as I need to connect DTR pin directly to reset).
* https://github.com/avrdudes/avrdude/pull/1309

MicroCore seems to have a problem that it does not pass the COM port number to avrdude.
```
C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1/avrdude.conf -v -pattiny13a -curclock -b57600 -Uflash:w:C:\Users\xfchen\AppData\Local\Temp\arduino_build_580486/Blink.ino.hex:i 

avrdude: Version 7.1-20230106 (b1da0b09)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : com1
         Using Programmer              : urclock
         Overriding Baud Rate          : 57600
avrdude ser_open() error: cannot open port \\.\com1: The system cannot find the file specified.

the selected serial port 
 does not exist or your board is not connected
avrdude main() error: unable to open programmer urclock on port com1

avrdude done.  Thank you.
```

Using command line and it works fine.
```
C:\work\avr\avrdude_test> C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1/avrdude.conf -v -pattiny13a -curclock -b57600 -P COM7 -Uflash:w:C:\Users\xfchen\AppData\Local\Temp\arduino_build_580486/Blink.ino.hex:i

avrdude: Version 7.1-20230106 (b1da0b09)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : COM7
         Using Programmer              : urclock
         Overriding Baud Rate          : 57600
         AVR Part                      : ATtiny13A
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
           flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATtiny13A

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
         delaying chip erase until first -U upload to flash
avrdude: reading input file C:\Users\xfchen\AppData\Local\Temp\arduino_build_580486/Blink.ino.hex for flash
         with 78 bytes in 1 section within [0, 0x4d]
         using 3 pages and 18 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 78 bytes flash ...

Writing | ################################################## | 100% 0.51 s

avrdude: 78 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Temp\arduino_build_580486/Blink.ino.hex

Reading | ################################################## | 100% 0.29 s

avrdude: 78 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2023-03-08:
1.2MHz works fine as well.
```
C:\work\avr\avrdude_test> C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1/avrdude.conf -v -pattiny13a -curclock -PCOM7 -b19200 -Uflash:w:C:\Users\xfchen\AppData\Local\Temp\arduino_build_841162/Blink_t13.ino.hex:i

avrdude: Version 7.1-20230106 (b1da0b09)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : COM7
         Using Programmer              : urclock
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny13A
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
           flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATtiny13A

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
         delaying chip erase until first -U upload to flash
avrdude: reading input file C:\Users\xfchen\AppData\Local\Temp\arduino_build_841162/Blink_t13.ino.hex for flash
         with 84 bytes in 1 section within [0, 0x53]
         using 3 pages and 12 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 84 bytes flash ...

Writing | ################################################## | 100% 0.85 s

avrdude: 84 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Temp\arduino_build_841162/Blink_t13.ino.hex

Reading | ################################################## | 100% 0.64 s

avrdude: 84 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2023-03-08:
I am not able to get 600KHz internal clock to work.

```
C:\work\avr\avrdude_test> C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\tools\avrdude\7.1-arduino.1/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1/avrdude.conf -v -pattiny13a -curclock -PCOM7 -b9600 -Uflash:w:C:\Users\xfchen\AppData\Local\Temp\arduino_build_841162/Blink_t13.ino.hex:i -B3

avrdude: Version 7.1-20230106 (b1da0b09)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : COM7
         Using Programmer              : urclock
         Overriding Baud Rate          : 9600
         Setting bit clk period        : 3.0
avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
avrdude urclock_getsync() warning: attempt 2 of 10: not in sync
avrdude urclock_getsync() warning: attempt 3 of 10: not in sync
avrdude urclock_getsync() warning: attempt 4 of 10: not in sync
avrdude urclock_getsync() warning: attempt 5 of 10: not in sync
avrdude urclock_getsync() warning: attempt 6 of 10: not in sync
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port COM7

avrdude done.  Thank you.

```

MCUdude, 2023-03-08:
> If 9m6 Hz @ 115200 baud works like a charm, then so should 4m8 Hz @ 57600 baud. It's even the same bootloader bit for bit, just that the entire "world" is slower. Will be the same for 1m2 Hz @ 14400 baud with exactly the same bootloader, no need to recompile or reflash. 

When running the [MicroCore oscillator calibration sketch](https://github.com/MCUdude/MicroCore/blob/master/avr/libraries/Serial_examples/examples/OscillatorCalibration/OscillatorCalibration.ino), it appears that the internal 4.8 MHz and the 9.6 MHz needs different OSCCAL values in order to be on point. The 4.8MHz one is more "off" than the 9.6 MHz one, but both of them need their OSCCAL value increased with about 5-8 steps. This is definitely why I'm having these issues.

I'm not sure it's possible to within the 256-byte mark, but it would be awesome if the bootloader could read EEPROM address 0 and load the value into the OSCCAL register if the value was appropriate. This way, the users could run the oscillator calibration sketch first, and then flash the bootloader. However, it would even be better if the OSCCAL value in EEPROM could be "burned" into the flash memory when the bootloader is flashed. Not sure how this could be accomplished. But the ATtiny13s are known for their inaccurate internal oscillators. I actually had an ATTiny13 I bought from Aliexpress where the 9.6MHz oscillator was actually running at 9 MHz. Pretty much useless for timing-critical applications unless the correct value was loaded into its OSCCAL register on boot.

> MicroCore seems to have a problem that it does not pass the COM port number to avrdude. My breakout board has two LEDs on PB3 and PB4.

Thanks for reporting! I've been using a `.avrduderc` file where a default serial port is specified, so I didn't notice this issue. I've just pushed a fix for this.

MCUdude, 2023-03-08:
> I am unhappy with the alphabetic order of the oscillator frequency directories in the urboot.hex github

Have a look at how I'll have to deal with the bootloader binary file names:
https://github.com/MCUdude/MicroCore/blob/master/avr/boards.txt

At least for me and my Arduino cores, it's important that the names are consistent and easy to handle with the limits the boards.txt parser has.

I'm not a huge fan of the [a-z] prefix because this would be difficult to deal with. It also means that existing hex files need to be renamed if/when new bootloaders are added to the directory.

I do agree that alphabetic order is very neat, as long as the boards.txt parser is able to handle the file names without too many workarounds

mcuee, 2023-03-08:
> > MicroCore seems to have a problem that it does not pass the COM port number to avrdude. My breakout board has two LEDs on PB3 and PB4.
> 
> Thanks for reporting! I've been using a `.avrduderc` file where a default serial port is specified, so I didn't notice this issue. I've just pushed a fix for this.

Yes, the latest git is good.

mcuee, 2023-03-08:
> When running the [MicroCore oscillator calibration sketch](https://github.com/MCUdude/MicroCore/blob/master/avr/libraries/Serial_examples/examples/OscillatorCalibration/OscillatorCalibration.ino), [...]

Yes. It is also mentioned in the datasheet Section 6.2.2 `Calibrated Internal 4.8/9.6 MHz Oscillator` and Section 173.

> 17.3 Calibration Bytes
> The signature area of the ATtiny13A contains two bytes of calibration data for the internal oscillator. The calibration data in the high byte of address 0x00 is for use with the oscillator set to 9.6
MHz operation. During reset, this byte is automatically written into the OSCCAL register to
ensure correct frequency of the oscillator.
> There is a separate calibration byte for the internal oscillator in 4.8 MHz mode of operation but
this data is not loaded automatically. The hardware always loads the 9.6 MHz calibration data
during reset. To use separate calibration data for the oscillator in 4.8 MHz mode the OSCCAL
register must be updated by firmware. The calibration data for 4.8 MHz operation is located in
the high byte at address 0x01 of the signature area.

> I'm not sure it's possible to within the 256-byte mark, but it would be awesome if the bootloader could read EEPROM address 0 and load the value into the OSCCAL register if the value was [...]

Good idea.

mcuee, 2023-03-08:
@MCUdude 

Another potential improvement for the MicroCore, As there is no urboot hex file for 128KHz internal oscillator, it is good to mention that in the menu. I am not so sure if that is easy to fix or not.

```
avrdude ioerror() OS error: file C:\Users\xfchen\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\2.2.1/bootloaders/urboot/fcpu_{bootloader.speed}/{bootloader.baudrate}_bps/urboot_attiny13a_{bootloader.speed}_{bootloader.baudrate}bps_swio_rxb0_txb1_ur_vbl.hex is not readable: No such file or directory
```

stefanrueger, 2023-03-09:
| would be awesome if the bootloader could read EEPROM address 0 and load the value into the OSCCAL register

Would it not make sense for the *calibration script* to store the OSCAL value? If the bootloader does that it renders EEPROM location 0 unusable for *all* applications and increases its size.

stefanrueger, 2023-03-09:
> Avrdude reads the EEPROM value (and perhaps a flag as an indicator that there's a valid OSCCAL value and not random, leftover data) and puts this straight into flash at a specific address so that the bootloader always loads this exact value into the OSCCAL register

That is the idea of template bootloaders (modify LED, WDTO, CS line for dual boot) in the bootloader .hex file at burn time. (Could be extended to OSCAL). Avrdude has no  (dedicated) support for bootloader burning but I am using a sketch on a ProMini to burn template bootloaders (not Avrdude)

stefanrueger, 2023-03-09:
From the [Attiny13a data sheet](https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/DataSheets/ATtiny13A-Data-Sheet-DS40002307A.pdf): To ensure stable operation of the MCU the calibration value should be changed in small steps. A variation in frequency of more than 2% from one cycle to the next can lead to unpredicatble behavior. Changes in OSCCAL should not exceed 0x20 for each calibration.

Is that a thing in reality? It would be quite some algorithm to having to read the reset-time factory calibration value, then compare to the destination OSCCAL and then increment/decrement its value until it matches the desired OSCCAL is reached. Do people do this or do they (brutally!) overwrite the OSCCAL register with any old value?

The more I think about it, the more I lean towards preparing bootloaders for several F_CPU spaced 1.5% or 2% apart.

MCUdude, 2023-03-12:
> To ensure stable operation of the MCU the calibration value should be changed in small steps. A variation in frequency of more than 2% from one cycle to the next can lead to unpredicatble behavior. Changes in OSCCAL should not exceed 0x20 for each calibration.
Is that a thing in reality?

I _highly_ doubt this, but in reality, the oscillator is usually only a few percent off anyways, so the "jump" isn't that big in reality.

> The more I think about it, the more I lean towards preparing bootloaders for several F_CPU spaced 1.5% or 2% apart.

This is a good idea, but IMO it's only needed for the ATtiny13, and for the 9.6 MHz and 1,2 MHz ("oscillator 1"), 4.8 MHz, and 600kHz ("oscillator 2"). And in all cases I've seen the oscillator has _always_ been too slow, and never too fast.

I'll probably provide a bootloader menu option like this with MicroCore:
```
- Bootloader
  - No
  - Yes (RX = PB1, TX = PB0, 0%)
  - Yes (RX = PB1, TX = PB0, -1.5%)
  - Yes (RX = PB1, TX = PB0, -3%)
  - Yes (RX = PB1, TX = PB0, -4.5%)
  - Yes (RX = PB1, TX = PB0, -6%)
  - Yes (RX = PB0, TX = PB1, 0%)
  - Yes (RX = PB0, TX = PB1, -1.5%)
  - Yes (RX = PB0, TX = PB1, -3%)
  - Yes (RX = PB0, TX = PB1, -4.5%)
  - Yes (RX = PB0, TX = PB1, -6%)
```

stefanrueger, 2023-03-13:
The other parts have internal oscillators, too, not? Normally at 8 MHz, so would make sense to provide a series of nearby frequencies for 8 MHz and 1 MHz for parts different from t13 and t13a. Also, most (all?) parts can be set to run on 128 kHz. We could create pre-compiled bootloaders for these frequencies, too.

We might only need to create these extra frequencies for parts *without* a UART. If the part has UART, one can always use an autobaud bootloader and use a  really *low* baudrate in relation to fcpu. fcpu/260 should work for sure. Higher baud rates may be in a region that is incompatible with fcpu, see these [example intervals](https://github.com/stefanrueger/urboot/blob/main/docs/makeoptions.md#serial-communication). Note I got the 260 from that discussion as 8*33/1.015.

BTW, there is a subtle challenge in stepping Fcpu up and down in intervals: The quantisation of high UART baud rates (relative to fcpu) for classic parts can be severely off in certain fcpu/br regions as seen above. Even if the UART (just about) works, eg, 8 MHz and 57,600 baud, the generated baud rate is +2.12% too high. This means that if I were to step down F_CPU by 1.5% I would still get the same BRR divisor (and therefore the same bootloader bit for bit) just that the nominal baud rate quantisation error drops to ~ +0.6%. *Soooooooo,* I am thinking to use SWIO when the UART baud rate quantisation error exceeds 0.75% and also not create SWIO bootloaders that exceed 0.75% quantisation error. Fun fact: for baud rates below fcpu/66.7 (and above some limit) my SWIO routines are *guaranteed* to be within 0.75% quantisation error: fcpu/br is the number of cycles per transmitted bit; here this is 66.7. The error in my SWIO routines is never more than 0.5 cycles, so max quantisation error is 0.5/66.7 = 0.75%. I believe optiboot has a 4.5% worst-case quantisation for baud rates in the vincinity of fcpu/66.7 because the code is based on the (for this suboptimal) Atmel AN 305 that allows the number of cycles per bit to be off by max 3. 

TLDR; hopefully my strategy will create bootloaders that are *different* for slightly different fcpu and that they cover the mis-tuned internal oscillators.

stefanrueger, 2023-03-13:
Just wondering why you plan to offer both RX = PB1, TX = PB0 and RX = PB0, TX = PB1? I thought your microcore had only the former pin assignment?

MCUdude, 2023-03-13:
> The other parts have internal oscillators, too, not?

Yes, they do, but I'm my experience, these are for some reason not as "off" as the one(s) found in the ATtiny13.

> We might only need to create these extra frequencies for parts without a UART.

That would make the most sense IMO. I'll only bundle Urboot autobaud bootloaders to my cores anyways, so a fine-tuned F_CPU value is in my case only needed with MicroCore/ATtiny13. But I understand that you target audiences other than just users of my Arduino cores.

> Just wondering why you plan to offer both RX = PB1, TX = PB0 and RX = PB0, TX = PB1? I thought your microcore had only the former pin assignment?

My plan is to offer both if the bootloader menu In Arduino IDE doesn't get too cluttered.

stefanrueger, 2023-03-13:
> fine-tuned F_CPU value is in my case only needed with MicroCore/ATtiny13

Understood. Will do, but like to do things in a general way (so they benefit a wider audience).

Will have a slightly different naming scheme in the next iteration for publishing pre-compiled bootloaders, so that browsing the tree is easier for a human.. Needed to deploy `+` instead of `_` so github sorts the directories in numerically consistent way. Could have used leading 0 but that turned out to look ugly and confusing. 

I am thinking of having `fcpu++8m0000_Hz` directory names for **external** oscillators (quartzes with 20 ppm and resonators with 2000 ppm accuracy, eg). These are good enough so that we can afford quantisation errors through the UART with up to 2.1% error. High baudrates are likely to work.

In contrast to this I propose to have `fint++8m0000_Hz` directory names for **internal** oscillators which are notoriously imprecise and which can exceed 2% deviation errors. For these I will limit generated bootloaders to small(ish) baud rates, say fcpu/67 and smaller for which I can generate baud rates with a *small* quantisation error and a few +/- percent varying fcpu.

While I write this, maybe, to be ultra-clear, I should name the directories for external oscillators `fext+..._hz` not `fcpu+..._hz`.

Also  making urprotocol=1 the default in file names, so the exception is `_stk500` and its absence means what used to be `_ur`. Hardware supported bootloaders will have `_hw` added and its absence means vector boot loader (what used to be indicated with `_vbl`). Am hoping that this settles in before I tag the urboot and urboot.hex projects with their "official" first release as v7.7.

[image]

[image]

[image]

stefanrueger, 2023-03-15:
OK, done. See how you like the new pre-compiled bootloaders in urboot.hex, @MCUdude 

Could you especially try the bootloaders for internal oscillators. They are meant to exhibit a low baud rate quantisation error. This and that there are a few ones for Fcpu variations should get the difficult 4.8 MHz b/loader going.

stefanrueger, 2023-03-15:
> As there is no urboot hex file for 128KHz internal oscillator, it is good to mention that in the menu.

There now @mcuee

MCUdude, 2023-03-15:
Thanks, @stefanrueger! I'll run some tests later this evening and update the bootloader binaries I've bundled with MicroCore.

IMO the new naming convention is a bit odd IMO. I do understand that all the pluses are there for padding reasons, but or look a bit cryptic to the naked eye. 

```
mcus/attiny13a/internal_oscillator/fint++0m133120_Hz/br++++0k3_bps/urboot+attiny13a++0m133120i++++0k3_swio_rxb0_txb1_led+b2.hex
```

stefanrueger, 2023-03-15:
> new naming convention

Yes, I needed to get used to that, too.

I tried underscores (sorts wrong), dashes (sorts wrong), dots (looks odd with proportional font), leading zeros (also looks odd, particularly baud rates are hard to read as they span 4 orders). Haven't tried spaces (they *are* a pain for shell scripts but apparently they sort wrong too) and haven't tried colons (these would be funny, particularly for windows :wink:) Any suggestions? Now is a good time to get this right.

Somehow Unix ls is remarkable in that ordinary ASCII sort is too much to ask. Github also doesn't give you much control over order. 

Do you have a suggestion that makes github display these files in the "right" order?

MCUdude, 2023-03-15:
> Do you have a suggestion that makes github display these files in the "right" order?

Not really, at least not something that ticks all the boxes.

Instead of flashing bootloaders that are compiled to be _slightly_ slower or faster than the built-in oscillator, I think I'll instead slightly increase or decrease the baud rate. The effect is the same, but it makes it way easier to make it work seamlessly in Arduino IDE. The returning problem is that all menu options are static, so if all menu options would be valid, I'll have to provide "slightly off" bootloaders for all the external oscillator options as well.

MCUdude, 2023-03-15:
After trying to get the new naming convention to work with Arduino IDE, I have some thoughts. The "older" style was easier to work with from an Arduino point of view. The bootloader was just more predictable, as there was no padding, but everything is possible, it's just a bit more work

* The folder names inside `external_oscillator`and `internal_oscillator` should have the same prefix. If not, it's just one more thing that needs to be taken care of when automizing the process of picking the correct bootloader. I also see no need for the `i` (for internal) and `x` (for external) in the file names. I believe there's no reason why a bootloader compiled for an 8 MHz external oscillator can't run using the internal one. It's just one more this that has to be dealt with.
* I personally find the padding so difficult to read compared to the old style, that I'd actually prefer a "non-sorted" list with names that would make sense. Everyting is nicely organized in folders anyways, so for instance:
```
attiny13a/internal_oscillator/fint++4m8000_Hz/br+++38k4_bps/urboot+attiny13a++4m8000i+++38k4_swio_rxb1_txb0_led+b2.hex
attiny13a/external_oscillator/fcpu+18m4320_Hz/br++125k0_bps/urboot+attiny13a+18m4320x++125k0_swio_rxb0_txb1_led+b2.hex 

attiny13a/internal_oscillator/4.8MHz/38400bps/urboot_attiny13a_4.8MHz_38400bps_swio_rxb1_txb0_led+b2.hex
attiny13a/external_oscillator/18.432MHz/115200bps/urboot_attiny13a_18.432MHz_115200bps_swio_rxb1_txb0_led+b2.hex 
```

stefanrueger, 2023-03-16:
Maybe we should distinguish between directory names and file names. The directory names should have padding to allow for numeric sorting in github. There can now be 30+ different fcpu and 30+ different baud rates.

A directory like [this](https://github.com/stefanrueger/urboot.hex/tree/main/mcus/atmega328p/external_oscillator) will be much easier to navigate than this order: fcpu_0.125MHz fcpu_0.2304MHz fcpu_0.25MHz fcpu_0.4608MHz fcpu_0.5MHz fcpu_0.9216MHz fcpu_10MHz fcpu_11.0592MHz fcpu_12MHz fcpu_1.3824MHz fcpu_14.7456MHz fcpu_1.5MHz fcpu_16MHz fcpu_1.8432MHz fcpu_18.432MHz fcpu_1MHz fcpu_20MHz fcpu_22.1184MHz fcpu_2.304MHz fcpu_24MHz  fcpu_2.5MHz fcpu_2.7648MHz fcpu_2MHz fcpu_3.6864MHz fcpu_3MHz fcpu_4.608MHz fcpu_4MHz fcpu_5.5296MHz fcpu_5MHz fcpu_6MHz fcpu_7.3728MHz fcpu_8MHz fcpu_9.216MHz

> see no need for the i (for internal) and x (for external) in the file names

In filenames this is as redundant as is the part name, fcpu and br. I want the filename to be unique, though: no two bootloaders in the `mcus` directory tree have the same file name. In theory you could put them all in a single directory.

'i' and 'x' bootloaders are *different*: 'i' ones have a quantisation error of max 0.5% and 'x' ones of max 2.1%. Allowing higher quantisation error potentially makes the code smaller (eg,  UART in classic parts with high baud rates have high quant error). 

The philosophy is that I want +/- 2.5%. External oscillators are *at worst* 0.2% off, so can afford to generate 2.1% quantisation error. Internal oscillators have a factor error of up to 2%, so want to force quant error to below 0.5%. Of course you can use either type for either oscillator. For bootloaders following a series fcpu-4%, fcpu-3%, ... fcpu+4% it  only makes sense to offer 'i' types: they cover a  frequency spectrum and are never farther than 2% from each other...

How about changing from

`cores/minicore/atmega328p/external_oscillator/fcpu+16m0000_Hz/br++115k2_bps/urboot+atmega328p+16m0000x++115k2_uart0_rxd0_txd1_ee_led+b5_fr_ce.hex`

to

`cores/MiniCore/ATmega328P/external_oscillator/fcpu+16.000000_MHz/br++115.200_kbps/urboot_m328p_16m0x115k2+uart0_rxd0_txd1_ee_led+b5_fr_ce.hex`

 - Camel case for MiniCore
 - Proper capitalisation for the part
 - `+` padding in directories for github sorting
 - `16.000000_MHz`: always 6 digits after decimal point for fcpu
 - `115.200_kbps`: always 3 digits after decimal point for br
 - internal_oscillator directory names like `fcpu++8.000000_MHz+1%`, not `fint++8m0800_Hz`
 - "coded" part, fcpu, i/x, br `m328p_16m0x115k2`: short and the same in each directory
 - `+` in filename to separate redundant part from parameters, can use wildcard: `fcpu+16.000000_MHz/br++115.200_kbps/*+uart0_rxd0_txd1_ee_led+b5_fr_ce.hex

MCUdude, 2023-03-16:
From a boards.txt point of view, it's actually better to have _one_ notation for the target name, f_cpu, and baud rate. If not, I'll have to deal with all these variants (see below) manually.

Lowercase notation for the AVR part number is much better, as this is already defined in boards.txt.

Instead of this notation
```
cores/MiniCore/ATmega328P/external_oscillator/fcpu+16.000000_MHz/br++115.200_kbps/urboot_m328p_16m0x115k2+uart0_rxd0_txd1_ee_led+b5_fr_ce.hex
```
boards.txt would be way easier to read and manage if it was like this instead.
```
cores/MiniCore/atmega328p/external_oscillator/fcpu+16.000000_MHz/br++115.200_kbps/urboot_atmega328p_16.000000_115.200_uart0_rxd0_txd1_ee_led+b5_fr_ce.hex
```
But the absolute best for Arduino use would be this (same way I named optiboot flash), but it doesn't sort all that well.
```
cores/MiniCore/atmega328p/external_oscillator/16000000L/115200/urboot_atmega328p_16000000L_115200_uart0_rxd0_txd1_ee_led+b5_fr_ce.hex
```

How to deal with the "second best option" to get the bootloader file name based on the settings in the IDE tools menu:
```
# build.mcu and upload.port are defined elsewhere in boards.txt as both of these are user selectable
328.menu.clock.16MHz_external=External 16 MHz
328.menu.clock.16MHz_external.urboot.type=external_oscillator
328.menu.clock.16MHz_external.upload.speed=115200
328.menu.clock.16MHz_external.urboot.speed=115.200
328.menu.clock.16MHz_external.build.f_cpu=16000000L
328.menu.clock.16MHz_external.utboot.clockspeed=16.000000

328.menu.bootloader.uart0.bootloader.file=urboot/{build.mcu}/{urboot.type}/fcpu+{urboot.clockspeed}_MHz/br++{urboot.speed}_kbps/urboot_{build.mcu}_{urboot.clockspeed}_{urboot.speed}_{upload.port}_rxd0_txd1_ee_led+b5_fr_ce.hex
```

How to deal with the best option:
```
328.menu.clock.16MHz_external=External 16 MHz
328.menu.clock.16MHz_external.urboot.type=external_oscillator
328.menu.clock.16MHz_external.upload.speed=115200
328.menu.clock.16MHz_external.build.f_cpu=16000000L

328.menu.bootloader.uart0.bootloader.file=urboot/{build.mcu}/{urboot.type}/{build.f_cpu}/{upload.speed}/urboot_{build.mcu}_{build.f_cpu}_{upload.speed}_{upload.port}_rxd0_txd1_ee_led+b5_fr_ce.hex
```

As you can see, the boards.txt file gets significantly less complicated in the second example, and I'll have to add the same fields for _every_ clock frequency option that's possible to select in Arduino IDE (328.menu.clock.20MHz_external, 328.menu.clock.18.432MHz_external)

stefanrueger, 2023-03-16:
Yes, there is always the tension between human readable and machine readable.

> better to have one notation for the target name, f_cpu, and baud rate

That's what is currently the case, not? You just dislike the visuals of the `+`, or?

Also, I like short names. How about

```
cores/MiniCore/atmega328p/external_oscillator/fcpu+16x0/br++115k2/urboot_atmega328p_16x0_115k2_uart0_rxd0_txd1_ee_led+b5_fr_ce.hex
```

`i` and `x` are important for the reasons outlined above. Think of them as tolerance bands as in resistors.

stefanrueger, 2023-03-16:
> padding

Can the  328.menu.bootloader.uart0.bootloader.file entry contain wildcards? Then you could use
`.../fcpu*+{urboot.clockspeed}/br*+{urboot.speed}/..`

I would generate the board entries with a program. Doing that by hand is punishing

MCUdude, 2023-03-16:
> Can the 328.menu.bootloader.uart0.bootloader.file entry contain wildcards? 

Sadly, no.

> I would generate the board entries with a program.

That would absolutely be the best, but to me, this is currently more work than implementing this manually in boards.txt.

However, if the padding plusses were used in the file name as well, I could take a shortcut:

```
build.mcu and upload.port are defined elsewhere in boards.txt as both of these are user selectable

328.menu.clock.16MHz_external=External 16 MHz
328.menu.clock.16MHz_external.urboot.type=external_oscillator
328.menu.clock.16MHz_external.upload.speed=115200
328.menu.clock.16MHz_external.urboot.speed=++115k2
328.menu.clock.16MHz_external.build.f_cpu=16000000L
328.menu.clock.16MHz_external.utboot.clockspeed=+16x0

328.menu.clock.8MHz_external=External 8 MHz
328.menu.clock.8MHz_external.urboot.type=external_oscillator
328.menu.clock.8MHz_external.upload.speed=57600
328.menu.clock.8MHz_external.urboot.speed=+++57k6
328.menu.clock.8MHz_external.build.f_cpu=8000000L
328.menu.clock.8MHz_external.utboot.clockspeed=++8x0

328.menu.bootloader.uart0.bootloader.file=urboot/{build.mcu}/{urboot.type}/fcpu{urboot.clockspeed}/br{urboot.speed}/urboot_{build.mcu}_{urboot.clockspeed}_{urboot.speed}_{upload.port}_rxd0_txd1_ee_led+b5_fr_ce.hex
```

stefanrueger, 2023-03-16:
Let me think about this over the weekend. I am sure I find a good solution.

MCUdude, 2023-03-16:
> This is one more difficulty unless you define upload.port as `uart0_rxd0_txd1`

Yes, I'll have to add `328.menu.bootloader.uart0.upload.port=uart0_rxd0_txd1`. instead of the current `328.menu.bootloader.uart0.upload.port=UART0`. Good idea. I'm only using upload.port to identify the UART number when the bootloader file is selected.

> Also to consider external_oscillator does not always use UARTs

Correct, but my plan is to only offer hardware UART bootloaders with MiniCore. MicroCore, however, will use `swio`.

stefanrueger, 2023-03-16:
> only offer hardware UART bootloaders with MiniCore

72 of the 370 fcpu/br combos don't use the UART b/c the quantisation error is too large, most famously 8 MHz/115,200 baud.

MCUdude, 2023-04-09:
The new names for the files and folder in the `cores` folder work great with Arduino IDE!
One thing though. Could you please provide -6% and -8% bootloaders for the ATtiny13 as well?
I have a chip here that won't work with the default 4.8MHz internal oscillator bootloader unless I use a baudrate of 4500 instead of 4800.

stefanrueger, 2023-04-10:
OK, a wider spread is needed than I imagined. How about the variation is in steps of 1.25% and has a range of 13 steps from -7.5% to 7.5%? Would that work? Coincidentally, 4500 = 4800-6.25%

MCUdude, 2023-04-10:
I have never seen an internal oscillator being faster than it should, only slower. And since UART can handle a few percent errors, it should be fine with 1.5 or 2% steps I think.

How about:  
104%, 102%, 100%, 98%, 96%, 94%, 92%, 90%

This should even cover the chips that "fell off the delivery truck" and sold on the black market.

MCUdude, 2023-04-10:
I read on a forum that some ATtiny13 chips are as bas as 10% off sometimes. Usually, the 4.8MHz one is the worst, because the default calibration value that gets automatically is for the 9.6 MHz one.

Which means that in order get the 4.8MHz oscillator to be consistent, we'll have to read the second byte in the calibration register using Avrdude, and ideally write this to a fixed address so that the bootloader can load this value into its OSCCAL register. This isn't currently possible using Avrdude 7.1, but with the feature suggested in https://github.com/avrdudes/avrdude/issues/1322, I think it should be _possible_. However, it's probably difficult to make ut fit within 256 bytes.

stefanrueger, 2023-04-10:
> never seen an internal oscillator being faster than it should, only slower

That's interesting! Could make boundaries asymmetric. Is that documented somewhere? Is it plausible owing to the waty factory calibration and/or the workings of the `OSCCAL` register.

[20 more replies]