stefanrueger/urboot #24: What are the benefits that this bootloader consumes less memory?

kr4fty, 2023-04-21:
Hello, sorry for the question. I am testing the new firmware on my TransistorTester (MCU: atmega324pa), but when I want to add more features to said tester, I cannot upload it because it tells me that the space is exceeded (>100%). For this reason I thought that by putting this bootloader and it taking up less space than the default bootloader, I could add more functionality to the firmware, but avrdude throws me a message that it already exceeded the limit and does not allow me to upload it to said chip.
I would have to do something else, did you want to remap the memory? what am i missing?

stefanrueger, 2023-04-21:
Do you actually use a bootloader to upload the software? In which case you would connect the target MCU/board directly to the PC with a serial cable or an USB to serial adapter; the MCU of your board (your IC tester) would have a small program (bootloader) to burn the program onto its *own* chip (_without_ an external programmer such as arduino ISP). 

If that is the case then urboot is a really small bootloader (ca 50% smaller than optiboot), and you can choose one that has only 256 bytes for the ATmega324PA MCU. This [directory](https://github.com/stefanrueger/urboot.hex/tree/main/mcus/atmega324pa/watchdog_2_s/autobaud/uart0_rxd0_txd1/no-led) has some that have 2 s WDTO, use UART 0, detect the baud rate of the host automatically, and do not blink an LED on use). But remember, the maximum program you can upload is 32768 - 256 bytes. **You would need to burn this bootloader first using an external programmer.** You may need to set fuses appropriately, see [here](https://github.com/stefanrueger/urboot/blob/main/README.md#usage). After you did that, you can upload directly from the PC to the IC tester board (you would need to use `avrdude -c urclock -xnometadata` for this, `avrdude -c arduino` does not work for that).

If you are really, really hard up for space, then you might want to burn your program without bootloaders. Then you have the full 32k available for your program. You will always need to use an intermediary device between the PC with AVRDUDE and the IC tester board with the ATmega324PA. Also you will need to set the fuses to jump to reset (instead of the boot section) in this case.

Has that answered your questions?

kr4fty, 2023-04-21:
So, if I wanted to save more space, should I use for example URBOOT and then record directly using the serial port? or do i understand wrong?

Maybe I could use this interface, for example?

<image src="https://2.bp.blogspot.com/_Ae0PmYTPcTs/SUlKFk7M2TI/AAAAAAAAALY/GlLLCO2AQ1c/s400/AVR.ISP.SCHEMATIC.png">

What would happen if I don't have a physical serial port and if I wanted to use a USB-TTL adapter?

I'm sorry if I'm asking and saying very basic things for you

stefanrueger, 2023-04-21:
You save the most space by not using a bootloader at all. Advantage: you can use all flash for your application program. Disadvantage: you always need to use an intermediary (physical) programmer, less good for rapid prototyping.

The normal, typical setup for bootloader usage is
 - Your board has an MCU with a bootloader already burned on it
 - You connect that board directly to a USB socket  of your PC
      + Either with a USB/serial converter cable (sometimes that has a little board with an FTDI IC or a CH431a etc)
      + Or your board *already* has a FTDI/CH431a/etc part on it and has a USB (often micro) socket
      + Or your PC has a serial interface with 5 V or 3.3 V (for example the Raspi has RX/TX on its 40 Pin GPIO header)
 - You can directly connect the board to the PC and you can see debug messages from the board to the PC
 - You can immediately program the board using AVRDUDE without using an intermediary physical programmer

Does your  TransistorTester (MCU: atmega324pa) currently have bootloader and do you use it that way? The answer to that question is no if you program using an intermediary physical programmer. In this case keep programming it that way and do not use a bootloader.

If your board has a bootloader and you already program it using that bootloader find out how big that bootloader is. That bootloader then is most likely between 512 bytes and 4096 bytes. The fuse setting BOOTSZ would tell you how big that bootloader is (and in order to read these you would need a physical programmer connected, say with the ISP header). If that is the case *and* if you want to keep programming your board via its own bootloader then you might want to look at _replacing the boards's bootloader_ with a 256-byte urboot vector bootloader. You would need to have access to the ISP header with the 4 SPI signals and Reset in order to replace the bootloader. Not all boards have that, and if your TransistorTester board lacks an ISP header you may be out of luck!

The schematic you have shown is unlikely to help/work. The connection between PC and a bootloader board ultimately needs to be from the PC to the RX/TX signals of the MCU.

kr4fty, 2023-04-21:
Ok, now the pieces of the puzzle are coming together. 

I thought the bootloader was essential to boot and load the program just like the bootloader and OS do on a PC. What I didn't understand before was that if a physical programmer is used, by ISP, now the bootloader is not needed ([reference](https://forum.arduino.cc/t/how-to-completely-remove-the-bootloader/172751/2)), if I'm not mistaken.

stefanrueger, 2023-04-21:
Correct. If you have a bootloader and want to remove it the easiest way is to ensure the BOOTRST fuse bit is set to 1 (on the ATmega324PA this should be bit 0 of the high fuse). That way, the MCU jumps to the ordinary reset vector on any reset (and not the selected HW supported boot section). You can do so with an external ISP programmer. Don't change the other bits of hfuse (otherwise you risk disabling ISP programming). 

And if ever you are tired of using a physical programmer in your development and want a slick uploading experience for rapid prototyping, give urboot a whirl. It's worth every byte :wink:

kr4fty, 2023-04-21:
Good perfect. I just realized that no matter how much I record a bootloader, it is deleted when uploading the firmware to the board (using ArduinoISP as programmer).

```
> avrdude -c arduino -P /dev/ttyUSB0 -p m324pa -b 115200 -U \
  flash:w:./ComponentTester.hex:a -U eeprom:w:./ComponentTester.eep:a
```

Thank you again for your patience and excellent explanation.

stefanrueger, 2023-04-21:
Sure, that will be the case. The *complete* flash memory will be deleted with your command line (including any previous bootloader that might have been there). Do check that 
```
avrdude -c arduino -P /dev/ttyUSB0 -p m324pa -b 115200 -U hfuse:r:-h
```
gives you a hex byte where the low bit is set. Best to double check with the ATmega324PA data sheet that the hfuse and the other fuses are set correctly

kr4fty, 2023-04-21:
```
> avrdude -c arduino -P /dev/ttyUSB0 -p m324pa -b 115200 -U hfuse:r                                                                                                           
avrdude error: invalid update specification
avrdude error: unable to parse update operation 'hfuse:r'
    ~/ComponentTester-1.49m   avrdude -c arduino -p m324pa -P /dev/ttyUSB0  -t                                                                                                                          1 ✘ 

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9511 (probably m324pa)
avrdude> dump hfuse

Reading | ################################################## | 100% 0.00 s 

0000  d9                                                |.               |
```

stefanrueger, 2023-04-22:
OK, good luck and all, closing the issue

closed 2023-04-22 by stefanrueger