stefanrueger/urboot #38: can't enter the bootloader after app flashed.

hlyi, 2024-06-24:
I burned the bootloader into a "pro mini" ( atmega328p runs at 3.3v with 8MHz crystal) with the following build configuration:
` make MCU=atmega328p F_CPU=8000000L AUTOBAUD=0 SWIO=1 RX=AtmelPC3 TX=AtmelPC2 BAUD_RATE=57600 WDTO=1S EEPROM=1 VBL=0 DUAL=0  BLINK=1 LED=AtmelPB5 LEDPOLARITY=1  EXITFE=2 QEXITEND=1 CHIP_ERASE=1 UPDATE_FL=4`

It worked first time without application code ( used reset button to enter the bootloader). After application code burned (using the bootloader). The device can not enter bootloader anymore (with reset button). Did I can configure something wrong?

Thanks

stefanrueger, 2024-06-24:
Sounds like the fuses still need to be set to jump to the bootloader on reset and to the correct bootloader size. With newer AVRDUDE versions you can do that by
```
$ avrdude -qq -c dryrun -p m328p -T "config bootsz=bs_256w; config bootrst=boot_section; config"
```
Obviously, you need to replace `dryrun` with the programmer that you used for uploading the bootloader (usbasp, tinyusb, ...).

hlyi, 2024-06-24:
Thanks @stefanrueger! for the quick reply. I am running avrdude 7.1 It doesn't have support for option T. I got the following error:
`avrdude: invalid option -- 'T'`.

Here is the fuse setting: LFUSE:0xff, HFUSE:0xdf, EFUSE:0xfd. Any of the them was incorrect?

closed 2024-07-03 by stefanrueger