stefanrueger/urboot #39: Support: Unable to write flash with bootloader on ATmega8

gauravahuja, 2024-07-07:
I burned the bootloader `urboot/atmega8/watchdog_1_s/autobaud/uart0_rxd0_txd1/led+b5/urboot_atmega8_pr_ee_ce.hex
` with lock and fuses as [lock:0xFF hfuse:0xC5 lfuse:0xBF](https://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega8&LOW=BF&HIGH=C5&LOCKBIT=FF)

The bootloader is running and I can enter into terminal mode as shown from my terminal session below. However, the bootloader fails to write to flash. I am not sure what the issue is. I am using a Apple M1 Pro with Sonoma 14.4.1.

Please help.
Thanks

```
avrdude -xdelay=100 -v -p atmega8 -C /Users/gaurav/.platformio/packages/tool-avrdude/avrdude.conf -c urclock -xnometadata -b 9600 -D -P /dev/cu.usbserial-1110 -t

avrdude: Version 7.3-20240428 (37ac6eba)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/gaurav/.platformio/packages/tool-avrdude/avrdude.conf
avrdude avr_get_mem_type() warning: avr_mem_order[] does not know data; add to array and recompile
         User configuration file is /Users/gaurav/.avrduderc
         User configuration file does not exist or is not a regular file, skipping
avrdude main() warning: System wide configuration file version ()
        does not match Avrdude build version (7.3-20240428 (37ac6eba))

         Using port            : /dev/cu.usbserial-1110
         Using programmer      : urclock
         Setting baud rate     : 9600
         AVR Part              : ATmega8
         Programming modes     : ISP, HVPP, SPM
         Programmer Type       : Urclock
         Description           : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATmega8
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8)
avrdude> dump eeprom 0 16
Reading | ################################################## | 100% 0.06 s
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> dump flash 0 32
Reading | ################################################## | 100% 0.00 s
0000  3f cf 64 c0 63 c0 62 c0  61 c0 60 c0 5f c0 5e c0  |?.d.c.b.a.`._.^.|
0010  5d c0 f1 c1 5b c0 5b c2  35 c2 58 c0 57 c0 56 c0  |]...[.[.5.X.W.V.|
avrdude> write flash .pio/build/Upload_UART/firmware.hex
Caching |                                                    | 0% 0.00 s
avrdude cmd_write() warning: (write) programmer write protects flash address 0x0000
avrdude cmd_write() warning: (write) programmer write protects flash address 0x0001
Caching | #                                                  | 1% 0.00 s
avrdude cmd_write() warning: (write) programmer write protects flash address 0x0024
Caching | ################################################## | 100% 1.97 s
avrdude> quit
avrdude: synching cache to device ...
Writing |                                                    | 0% 0.00 s
avrdude urclock_recv() warning: programmer is not responding
Writing | -------------------------------------------------- | 0% 0.50 s
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash
avrdude writeCachePage() error: flash access error at addr 0x00c0
avrdude: synching cache to device ...
Writing | -------------------------------------------------- | 0% 0.33 s
avrdude urclock_res_check() error: protocol expects sync byte 0xa0 but got 0x00 in urclock_paged_write()
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash
avrdude writeCachePage() error: flash access error at addr 0x00c0
avrdude urclock_recv() warning: programmer is not responding

avrdude done.  Thank you.
```

stefanrueger, 2024-07-07:
AVRDUDE should be given the conf file that matches its version. An older conf file is no good.

If you want to upload a hex file to the target, it's better to use `-U` and not the terminal. Have you tried sth like
```
avrdude -xdelay=100 -v -p atmega8  -c urclock -xnometadata -b 9600 -D -P /dev/cu.usbserial-1110 -U .pio/build/Upload_UART/firmware.hex
```

The reason being that vector bootloaders need to patch the uploaded application, which is done on `-U`.

gauravahuja, 2024-07-08:
@stefanrueger I tried the following command as you suggested. I am also using the avrdude that was installed along with platformio instead of the one I had installed with homebrew, so the config file should match the version. Unfortunately it still fails to write.

`/Users/gaurav/.platformio/packages/tool-avrdude/avrdude -xdelay=100 -v -p atmega8 -C /Users/gaurav/.platformio/packages/tool-avrdude/avrdude.conf -c urclock -xnometadata -b 9600 -D -P /dev/cu.usbserial-110 -U flash:w:.pio/build/Upload_UART/firmware.hex:i`

```
avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/gaurav/.platformio/packages/tool-avrdude/avrdude.conf
         User configuration file is /Users/gaurav/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-110
         Using Programmer              : urclock
         Overriding Baud Rate          : 9600
         AVR Part                      : ATmega8
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash                  33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATmega8
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8)

avrdude: processing -U flash:w:.pio/build/Upload_UART/firmware.hex:i
avrdude: reading input file .pio/build/Upload_UART/firmware.hex for flash
         with 1862 bytes in 1 section within [0, 0x745]
         using 30 pages and 58 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 1862 bytes flash ...
Writing | 
avrdude urclock_recv() warning: programmer is not responding
#------------------------------------------------- | 3% 0.50s
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
.
. Same error about 4000 times
.
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash 
 ***failed;
avrdude: 1862 bytes of flash written
avrdude: verifying flash memory against .pio/build/Upload_UART/firmware.hex
Reading | -------------------------------------------------- | 0% 0.21s
avrdude urclock_res_check() error: protocol expects sync byte 0xa0 but got 0x00 in urclock_paged_load()
avrdude urclock_recv() warning: programmer is not responding
avrdude avr_read_mem() error: unable to read byte at address 0x0000
avrdude avr_read_mem() error: read operation not supported for memory flash
avrdude do_op() error: unable to read all of flash memory, rc=-2
avrdude urclock_recv() warning: programmer is not responding
avrdude ser_close() OS error: cannot reset attributes for device: Invalid argument

avrdude done.  Thank you.
```

stefanrueger, 2024-07-09:
Beats me. Fuses are OK. You can communicate with the b/loader. I predict `avrdude -c urclock -xshowall` will say something sensible. Maybe try higher/lower baudrates? Your fuses indicate an external full-rail-to-rail, high-frequency XTAL, so  this autobaud b/loader should be able to deal with much higher baud rates. Maybe try a new urboot v8.0 bootloader? This one should be the equivalent version of yours:

```
wget https://raw.githubusercontent.com/stefanrueger/urboot.hex/main/mcus/atmega8/watchdog_1_s/autobaud/uart0_rxd0_txd1/led%2Bb5/urboot_m8_1s_autobaud_uart0_rxd0_txd1_led%2Bb5_pr_ee_ce.hex
```

It should be the penultimate one of the table in this [directory](https://github.com/stefanrueger/urboot.hex/tree/main/mcus/atmega8/watchdog_1_s/autobaud/uart0_rxd0_txd1/led%2Bb5)

v8.0 bootloaders have been tested on related parts ATmega16 and ATmega32A. Other than the trouble shooting list of the [urboot](https://github.com/stefanrueger/urboot) project (scroll down) I have no idea what could have gone wrong. I did have problems in the past with some MCUs, which were cheapo knock-off versions of the real thing but with subtle differences. I don't expect this to happen frequently, though.

stefanrueger, 2024-07-13:
@gauravahuja Have you made progress with your issue? Let me know once you've solved this.

I don't have the ATmega8, but I've just tested urboot v8.0 myself with the bigger sibling ATmega16A and it [worked just fine](https://github.com/stefanrueger/urboot/discussions/36#discussioncomment-10036669).

gauravahuja, 2024-07-13:
@stefanrueger, I was trying to source a different computer to test the setup. I was able to use a Linux machine and the hex file was successfully flashed to the ATmega8 with my existing setup. For some reason it always fails with my 14'' M1 MacBook Pro.

stefanrueger, 2024-07-13:
> For some reason it always fails with my 14'' M1 MacBook Pro

Maybe try the [build tips for Mac](https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS) here? If you are unable to get AVRDUDE to work with your urboot bootloader on your Mac, please raise an [issue at the avrdude project](https://github.com/avrdudes/avrdude/issues).

Seeing this does not appear to be an urboot problem, I'm closing the issue.

Out of interest, so I can record a report of urboot working on this chip: is your ATmega8 bootloader v7.7 or v8.0? And is the actual chip an ATmega8 or an ATmega8A?

closed 2024-07-13 by stefanrueger

gauravahuja, 2024-07-13:
The chip is ATmega8L with urboot v7.7.1